stereolabs/zed-sdk #349: zed安装遇到问题，希望大佬们帮帮我.

xiaohuozi-code, 2021-03-05:
请问安装zed，运行ZED_Dignosizic,诊断相机连接时：processor
报错：unable to identify your processor,make sure that it fits the requirements；
并且cametest：camera not detected

john-lihn, 2021-03-11:
您方便貼放相關網址嗎? 方便查看原碼

xiaohuozi-code, 2021-03-11:
可以，现在问题是这个，运行zed/tools/ 下的各种可执行文件，报错：
[image]
试问如何解决运行zed的问题

xiaohuozi-code, 2021-03-11:
现在下载zed-ros-wrappe编译遇到问题：主要是报错（几个变量不是sl的成员？我初学代码，还不太懂如何解决，希望您能告知一二，万分感谢）
[ 97%] Building CXX object zed-ros-wrapper/zed_nodelets/CMakeFiles/ZEDNodelets.dir/src/zed_nodelet/src/zed_wrapper_nodelet.cpp.o
/home/qupengtao/catkin_ws/src/zed-ros-wrapper/zed_nodelets/src/tools/src/sl_tools.cpp: In function ‘void sl_tools::imageToROSmsg(sensor_msgs::ImagePtr, sl::Mat, std::__cxx11::string, ros::Time)’:
/home/qupengtao/catkin_ws/src/zed-ros-wrapper/zed_nodelets/src/tools/src/sl_tools.cpp:240:24: error: ‘U16_C1’ is not a member of ‘sl::MAT_TYPE’
     case sl::MAT_TYPE::U16_C1: /**< unsigned short 1 channel.*/
                        ^~~~~~
/home/qupengtao/catkin_ws/src/zed-ros-wrapper/zed_nodelets/src/tools/src/sl_tools.cpp:242:63: error: ‘ushort1’ is not a member of ‘sl’
       memcpy((uint16_t*)(&imgMsgPtr->data[0]), img.getPtr<sl::ushort1>(), size);
                                                               ^~~~~~~
/home/qupengtao/catkin_ws/src/zed-ros-wrapper/zed_nodelets/src/tools/src/sl_tools.cpp:242:63: note: suggested alternative: ‘uchar1’
       memcpy((uint16_t*)(&imgMsgPtr->data[0]), img.getPtr<sl::ushort1>(), size);
                                                               ^~~~~~~
                                                               uchar1
/home/qupengtao/catkin_ws/src/zed-ros-wrapper/zed_nodelets/src/tools/src/sl_tools.cpp:242:63: error: ‘ushort1’ is not a member of ‘sl’
/home/qupengtao/catkin_ws/src/zed-ros-wrapper/zed_nodelets/src/tools/src/sl_tools.cpp:242:63: note: suggested alternative: ‘uchar1’
       memcpy((uint16_t*)(&imgMsgPtr->data[0]), img.getPtr<sl::ushort1>(), size);
                                                               ^~~~~~~
                                                               uchar1
/home/qupengtao/catkin_ws/src/zed-ros-wrapper/zed_nodelets/src/tools/src/sl_tools.cpp:242:72: error: no matching function for call to ‘sl::Mat::getPtr<<expression error> >()’
      memcpy((uint16_t*)(&imgMsgPtr->data[0]), img.getPtr<sl::ushort1>(), size);
                                                                       ^
In file included from /home/qupengtao/catkin_ws/src/zed-ros-wrapper/zed_nodelets/src/tools/include/sl_tools.h:26:0,
                 from /home/qupengtao/catkin_ws/src/zed-ros-wrapper/zed_nodelets/src/tools/src/sl_tools.cpp:21:
/usr/local/zed/include/sl/Camera.hpp:2537:12: note: candidate: template<class N> N* sl::Mat::getPtr(sl::MEM) const
         N *getPtr(MEM memory_type = MEM::CPU) const;
            ^~~~~~
/usr/local/zed/include/sl/Camera.hpp:2537:12: note:   template argument deduction/substitution failed:
/home/qupengtao/catkin_ws/src/zed-ros-wrapper/zed_nodelets/src/tools/src/sl_tools.cpp:242:72: error: template argument 1 is invalid
      memcpy((uint16_t*)(&imgMsgPtr->data[0]), img.getPtr<sl::ushort1>(), size);
                                                                       ^
zed-ros-wrapper/zed_nodelets/CMakeFiles/ZEDNodelets.dir/build.make:62: recipe for target 'zed-ros-wrapper/zed_nodelets/CMakeFiles/ZEDNodelets.dir/src/tools/src/sl_tools.cpp.o' failed
make[2]: *** [zed-ros-wrapper/zed_nodelets/CMakeFiles/ZEDNodelets.dir/src/tools/src/sl_tools.cpp.o] Error 1
make[2]: *** 正在等待未完成的任务....
In file included from /home/qupengtao/catkin_ws/src/zed-ros-wrapper/zed_nodelets/src/zed_nodelet/src/zed_wrapper_nodelet.cpp:21:0:
/home/qupengtao/catkin_ws/src/zed-ros-wrapper/zed_nodelets/src/zed_nodelet/include/zed_wrapper_nodelet.hpp:699:7: error: ‘DETECTION_MODEL’ in namespace ‘sl’ does not name a type
   sl::DETECTION_MODEL mObjDetModel = sl::DETECTION_MODEL::MULTI_CLASS_BOX;
       ^~~~~~~~~~~~~~~
/home/qupengtao/catkin_ws/src/zed-ros-wrapper/zed_nodelets/src/zed_nodelet/src/zed_wrapper_nodelet.cpp: In member function ‘void zed_nodelets::ZEDWrapperNodelet::readParameters()’:
/home/qupengtao/catkin_ws/src/zed-ros-wrapper/zed_nodelets/src/zed_nodelet/src/zed_wrapper_nodelet.cpp:796:52: error: ‘sl::DETECTION_MODEL’ has not been declared
     if (model < 0 || model >= static_cast<int>(sl::DETECTION_MODEL::LAST))
                                                    ^~~~~~~~~~~~~~~
/home/qupengtao/catkin_ws/src/zed-ros-wrapper/zed_nodelets/src/zed_nodelet/src/zed_wrapper_nodelet.cpp:799:32: error: ‘mObjDetModel’ was not declared in this scope
       model = static_cast<int>(mObjDetModel);
                                ^~~~~~~~~~~~
/home/qupengtao/catkin_ws/src/zed-ros-wrapper/zed_nodelets/src/zed_nodelet/src/zed_wrapper_nodelet.cpp:799:32: note: suggested alternative: ‘mObjDetMutex’
       model = static_cast<int>(mObjDetModel);
                                ^~~~~~~~~~~~
                                mObjDetMutex
/home/qupengtao/catkin_ws/src/zed-ros-wrapper/zed_nodelets/src/zed_nodelet/src/zed_wrapper_nodelet.cpp:801:5: error: ‘mObjDetModel’ was not declared in this scope
     mObjDetModel = static_cast<sl::DETECTION_MODEL>(model);
     ^~~~~~~~~~~~
/home/qupengtao/catkin_ws/src/zed-ros-wrapper/zed_nodelets/src/zed_nodelet/src/zed_wrapper_nodelet.cpp:801:5: note: suggested alternative: ‘mObjDetMutex’
     mObjDetModel = static_cast<sl::DETECTION_MODEL>(model);
     ^~~~~~~~~~~~
     mObjDetMutex
/home/qupengtao/catkin_ws/src/zed-ros-wrapper/zed_nodelets/src/zed_nodelet/src/zed_wrapper_nodelet.cpp:801:36: error: ‘DETECTION_MODEL’ in namespace ‘sl’ does not name a type
     mObjDetModel = static_cast<sl::DETECTION_MODEL>(model);
                                    ^~~~~~~~~~~~~~~
[... cut ...]
zed-ros-wrapper/zed_nodelets/CMakeFiles/ZEDNodelets.dir/build.make:86: recipe for target 'zed-ros-wrapper/zed_nodelets/CMakeFiles/ZEDNodelets.dir/src/zed_nodelet/src/zed_wrapper_nodelet.cpp.o' failed
make[2]: *** [zed-ros-wrapper/zed_nodelets/CMakeFiles/ZEDNodelets.dir/src/zed_nodelet/src/zed_wrapper_nodelet.cpp.o] Error 1
CMakeFiles/Makefile2:2394: recipe for target 'zed-ros-wrapper/zed_nodelets/CMakeFiles/ZEDNodelets.dir/all' failed
make[1]: *** [zed-ros-wrapper/zed_nodelets/CMakeFiles/ZEDNodelets.dir/all] Error 2
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j8 -l8" failed

john-lihn, 2021-03-11:
你make編譯沒成功 試用 
> make -j8 l8

john-lihn, 2021-03-11:
> 你make編譯沒成功 試用
> 
> > make -j8 l8

或是 試用python 進行編譯運作