stereolabs/zed-sdk #350: Tracking custom objects

Zumbalamambo, 2021-03-09:
How do I use positional tracking for the custom objects that I have obtained by training some model on my own dataset?

obraun-sl, 2021-03-12:
Hi,
Do you mean the positional tracking of the camera? If yes, if the camera moves, you need to apply the pose to the position of the objects detected so that it goes from camera frame to world frame.
Another solution is to set the runtimeparameters::measure3D_reference_frame=REFERENCE_FRAME::WORLD, so that the depth/measure are given in world coordinate. Then Use the depth/point cloud to calculate the position of your detected objects in the world frame.

Zumbalamambo, 2021-03-12:
> Hi,
>
> Do you mean the positional tracking of the camera? If yes, if the camera moves, you need to apply the pose to the position of the objects detected so that it goes from camera frame to world [...]

Thank you! But I meant the positional tracking of the objects that are moving in the FOV of the camera... similar to the person/vehicle tracking but replacing them with the custom objects...

obraun-sl, 2021-03-13:
I see. That's not possible yet inside the sdk, but I mark this a feature request.

closed 2022-04-26 by github-actions[bot]