stereolabs/zed-sdk #351: ZEDfu failed to start

SimonSongg, 2021-03-10:
Platform: Jetson Xavier NX Jetpack 4.4.1 ZED sdk 3.4.1. I used the ZEDfu in the SDK tools and I can't open the live mode. Once I click the live bottom, ZEDfu exits without any error.

1chi217, 2021-03-18:
I have the same problem:
Jetson Xavier NX
Jetpack 4.4.1
ZED SDK 3.4.2
CUDA 10.2

Running ZEDfu over command line gives:
./ZEDfu 
cudaErrorInvalidValue
[1]    10912 segmentation fault (core dumped)  ./ZEDfu

obraun-sl, 2021-03-18:
Hi,
The crash has been reproduced and fix will be done for next release .

glucas350, 2021-03-19:
When is the next release planned? Is there a way to build early versions from source?

obraun-sl, 2021-03-27:
ZED SDK 3.5 is planned for next week.
You can follow @Stereolabs3D on Twitter to be notified as soon as it will be available for download

Note that you can also use the spatial mapping example provided which is similar to ZEDfu.
https://github.com/stereolabs/zed-examples/tree/master/spatial%20mapping

closed 2022-04-26 by github-actions[bot]