stereolabs/zed-sdk #352: AttributeError: 'pyzed.sl.ObjectDetectionRuntimeParameters' object has no attribute 'object_class_filte'

john-lihn, 2021-03-11:
@P-yver  I ran the example https://github.com/stereolabs/zed-examples/tree/master/tutorials/tutorial%206%20-%20object%20detection/python

error occurred AttributeError: 'pyzed.sl.ObjectDetectionRuntimeParameters' object has no attribute 'object_class_filte' 
How to solve it?

john-lihn, 2021-03-11:
> @P-yver I ran the example https://github.com/stereolabs/zed-examples/tree/master/tutorials/tutorial%206%20-%20object%20detection/python

[image]

> 
> error occurred AttributeError: 'pyzed.sl.ObjectDetectionRuntimeParameters' object has no attribute 'object_class_filte'
> How to solve it?

john-lihn, 2021-03-11:
> > @P-yver I ran the example https://github.com/stereolabs/zed-examples/tree/master/tutorials/tutorial%206%20-%20object%20detection/python
> 
> > error occurred AttributeError: 'pyzed.sl.ObjectDetectionRuntimeParameters' object has no attribute 'object_class_filte'
> > How to solve it?
>
>zed version
[image]

qt-truong, 2021-03-11:
Hi @john-lihn,

You are using the ZED SDK 3.2, and the `object_class_filter` attribute has been added to the Python API in ZED SDK 3.4.
You can download a more recent version of the SDK here : https://www.stereolabs.com/developers/release/

john-lihn, 2021-03-11:
Okay thank you, I will try it out, thank you

john-lihn, 2021-03-11:
Will you use the tensorflow h5 model or the pytorch model to perform the object detection function in the future, so that more experiments can be carried out in the future?

john-lihn, 2021-03-11:
[image]
Sorry to interrupt again, I have updated the latest version of python API, but this error still exists, how can I improve it?thank you

john-lihn, 2021-03-11:
[image]
Sorry to interrupt again, I have updated the latest version of python API, but this error still exists, how can I improve it, thank you

qt-truong, 2021-03-11:
There is a typo in your code : the attibute is "object_class_file**r**"

john-lihn, 2021-03-11:
Thank you very much for your help, it is ready to run now, thank you again

closed 2021-03-12 by john-lihn