stereolabs/zed-sdk #353: Body Tracking Example Code doesnt work

accolypse1994, 2021-03-11:
Hello,
The example code is working on my friend computer however the same code is not working in my computer. I am using PyCharm Community Edition 2020.2 and python 3.6. I got error viewer part of the code and it says "line 490, in init
    glutCreateWindow("ZED Body Tracking"). The image of the error has been attached. Could you help me please ? 
[image]
"

qt-truong, 2021-03-12:
Hello,

Can you try changing the call to `glutCreateWindow("ZED Body Tracking")` to `glutCreateWindow(b"ZED Body Tracking")` and see if that fixes your problem ?
It seems that there are issues on how strings are encoded as suggested here : https://stackoverflow.com/questions/27093037/python-glutcreatewindow-error-wrong-type

accolypse1994, 2021-03-12:
Thank you so much. It works after this.

closed 2021-03-12 by qt-truong