stereolabs/zed-sdk #354: Ask a question about object detection

john-lihn, 2021-03-12:
I want to ask, does the person detection of physical inspection need to use the tensorflow module or pytorch module to configure the person before it can run?
Or is it a model that has been configured internally so that everyone can recognize it?

obraun-sl, 2021-03-12:
Hi,
The models are already configured and ready to use without TFlow or pytorch.

john-lihn, 2021-03-12:
But I photographed myself with ZED2, he couldn’t recognize it, please ask him if he enters the weight file

obraun-sl, 2021-03-12:
The weight are already included in the model loaded by the sdk.
Can you post your code ?
What OS/PC are you using?

john-lihn, 2021-03-12:
I am testing zed-examples:
https://github.com/stereolabs/zed-examples/tree/master/object%20detection/image%20viewer/python

obraun-sl, 2021-03-12:
Try to reduce the detection threshold here : 
https://github.com/stereolabs/zed-examples/blob/master/object%20detection/image%20viewer/python/object_detection_image_viewer.py#L72
(something around 30).

Note that the camera won't detect objects under 50cm as well. You might be a little bit too close to the camera.

john-lihn, 2021-03-12:
Thank you very much, I am trying to see, thank you

closed 2021-03-15 by john-lihn