stereolabs/zed-sdk #358: Error: import pyzed.sl as sl

V1ctorrr, 2021-03-29:
I have finished the installation of "get_python_api.py",  but when I follow the tutorial "import pyzed.sl as sl", it says no module named "sl". Could you tell me how to figure out this error?

qt-truong, 2021-03-29:
Hello @V1ctorrr,

In order to help you out with your issue, could you please provide us with the following information :

- What OS, SDK version, Python version are you using ?
- What output log do you get when running the script `get_python_api.py` ? 
- Are you using a virtual environment ? (if that is the case, make sure to install the API **inside** your virtual environment)

V1ctorrr, 2021-03-29:
> Hello @V1ctorrr,
>
> In order to help you out with your issue, could you please provide us with the following information :
>
> * What OS, SDK version, Python version are you using ?
> * What output log do [...]

My OS is windows10, python version is 3.8; 
and the output is following:
"Installing collected packages: pyzed
Successfully installed pyzed-3.4
Done"
. Besides, I install this in the base environment of anaconda, and it shown in "D:\Soft\Users\Admin\anaconda3\Lib\site-packages"

qt-truong, 2021-03-29:
Thanks for the information. It seems that the installation went well ...
Are you using and IDE to run the sample ? If so, make sure to configure it for your virtual environment. There is a documentation page explaining how to do so here : [link](https://www.stereolabs.com/docs/app-development/python/virtual_env/)
If you are running your python script inside a terminal, make sure you have activated your anaconda environment before running your python script

V1ctorrr, 2021-03-29:
> Thanks for the information. It seems that the installation went well ...
> Are you using and IDE to run the sample ? If so, make sure to configure it for your virtual environment. There is a [...]

Thanks for your help. I have followed the doc you gave and run the script "Hello ZED.py" on Pycharm but still have the error. It looks strange that I got the output with my device ID.

V1ctorrr, 2021-03-31:
> You can try to run
> `conda list pyzed`
> Check if you have a pyzed package
> 
> If you don't have this package, you can run
> `python3 -m pip install pyzed-3.1-cp37-cp37m-linux_x86_64.whl`
> To add a package

It shows I have the package with version 3.4 and `body_tracking.py` also works even with this error.

john-lihn, 2021-03-31:
Excuse me, do you take a screenshot, the error code

V1ctorrr, 2021-03-31:
> Excuse me, do you take a screenshot, the error code

[image]

In the second line.

john-lihn, 2021-03-31:
尼是否嘗試直接運行

V1ctorrr, 2021-03-31:
> 您是否嘗試直接運行

尝试了，可以实时显示人体位姿。我尝试了读取IMU数据也可以输出，所以感觉很奇怪

john-lihn, 2021-03-31:
[image]
那您的問題應該是, 以vs code來講 他有所謂當前執行路徑, 若沒改過會是在環境外, 因此以環境外去查看程式, 會發現錯誤訊息

V1ctorrr, 2021-03-31:
> [image]
> 那您的問題應該是, 以vs code來講 他有所謂當前執行路徑, 若沒改過會是在環境外, 因此以環境外去查看程式, 會發現錯誤訊息

应该怎么设置呢，我这样尝试并不行
[image]

V1ctorrr, 2021-03-31:
> [image]
> 那您的問題應該是, 以vs code來講 他有所謂當前執行路徑, 若沒改過會是在環境外, 因此以環境外去查看程式, 會發現錯誤訊息

似乎没问题了，我是这样设置的。非常感谢~  
[image]

fakermmp, 2023-01-14:
想问您一下我也遇到这个问题但是我就是运行不了

fakermmp, 2023-01-14:
"C:\Program Files\Python310\python.exe" "C:\Program Files (x86)\ZED SDK\samples\body tracking\python\body_tracking.py" 
Running Body Tracking sample ... Press 'q' to quit

进程已结束,退出代码1

fakermmp, 2023-01-14:
我也是import pyzed.sl as sl      sl那里标红但是我运行不了