stereolabs/zed-sdk #359: Required memory for Object detection optimization is too high

thomascp, 2021-03-29:
Hi,

I am using `ZED-2` board, SDK is `ZED_SDK_Ubuntu20_full_cuda11.1_v3.4.2.run`.
My laptop graphic card is `NVIDIA Corporation GM108M [GeForce MX130]`.

When I call `ZED_Body_Tracking_Viewer`, it fails with `Required memory for Object detection optimization is too high for the current free memory` error, then it also reminds me that the corrupted SDK detected.
```
$ ./body\ tracking/cpp/ZED_Body_Tracking_Viewer 

[ZED][Object Detection] Please wait while the AI model is being optimized for your graphics card
 This operation will be run only once and may take a few minutes  |[ZED][Object Detection] Required memory for Object detection optimization is too high for the current free memory (630MB vs 562MB)
[ZED][Object Detection] Required memory for Object detection optimization is too high for the current free memory (630MB vs 562MB)
 Done[ZED] [Object Detection] MODEL_NOT_FOUND
[Sample][Error] enable Object Detection  | CORRUPTED SDK INSTALLATION : Corrupted SDK detected. Please reinstall the ZED SDK. 
Exit program.
```

I have reinstall SDK several times, so couldn't be this reason.
And I also checked output of `nvidia-smi`, the memory part is `375MiB /  2004MiB`.

So, if I am correct, my graphic card is not powerful enough to do body tracking? Right?

obraun-sl, 2021-03-30:
Hi,
Indeed, you have 2Gb of GPU memory (2004MiB) which is under the 3GB recommendation.

However, the samples are by default setup in maximum quality preset. You can change that to see if it still can fit your 2Gb : 
- Change the depth mode to PERFORMANCE here : 
https://github.com/stereolabs/zed-examples/blob/master/body%20tracking/cpp/src/main.cpp#L52

- Change the body tracking model to FAST mode here : 
https://github.com/stereolabs/zed-examples/blob/master/body%20tracking/cpp/src/main.cpp#L80

(Basically, setup as if you were on jetson mode...)

thomascp, 2022-04-20:
Sorry, forgot this issue, but I don't have a ZED-2 camera anymore, so close this.

closed 2022-04-20 by thomascp