stereolabs/zed-sdk #361: streaming failed. [UDP] Multicast group join failed (setsockopt())

reginalluna, 2021-04-02:
### sender device info:

1. System: Windows 10
2. GPU:  GTX1070
3. CUDA: 11.1
4. ZED SDK: 3.4.2
5. OpenCV (yes)

### receiver device info:
1. System: Ubuntu18.04
2. GPU: GTX750M
3. CUDA: 10.0
4. OpenCV (yes)

---
When I tried to run `python streaming_sender.py`, it shows 
[image]

And, I ran `./ZED_Streaming_`, it said:
`
`

reginalluna, 2021-04-02:
> ### sender device info:
> 
>     1. System: Windows 10
> 
>     2. GPU:  GTX1070
> 
>     3. CUDA: 11.1
> 
>     4. ZED SDK: 3.4.2
> 
>     5. OpenCV (yes)
> 
> 
> ### receiver device info:
> 
>     1. System: Ubuntu18.04
> 
>     2. GPU: GTX750M
> 
>     3. CUDA: 10.0
> 
>     4. OpenCV (yes)
> 
> 
> When I tried to run `python streaming_sender.py`, it shows
> [image]
> 
> And, I ran `./ZED_Streaming_`, it said:
> `Opening the stream requires the IP of the sender
Usage : ./ZED_Streaming_Receiver IP:[port]
You can specify it now, then press ENTER, 'IP:[port]': 224.0.0.0:30000
[ZED][Streaming] Error: Stream opening timeout
[Sample][Error] Camera Open  | FAILURE : The function does not perform as expected. Exit program.
 `

obraun-sl, 2021-04-03:
The IP adress you enter does not seem to be correct as it lloks like a multicast adress.
You need to enter the IP adress of the sender (the streaming is based on RTP unicast). 
To get the IP of the sender, type ipconfig on your windows PC, and you should get a IPv4 adress under the form 192.168.X.X

reginalluna, 2021-04-03:
> The IP adress you enter does not seem to be correct as it lloks like a multicast adress.
> You need to enter the IP adress of the sender (the streaming is based on RTP unicast).
> To get the IP of the sender, type ipconfig on your windows PC, and you should get a IPv4 adress under the form 192.168.X.X

Oh, let me try it again. Thank you. And, there's a weird problem that happened to my PC. The ZED camera has no reaction (not working) toward the USB3.0 ports of the PC after I unplugged it from my PC to my Ubuntu laptop and replugged it back to the PC.

reginalluna, 2021-04-03:
> ipconfig

Hi, while I ran `ipconfig`, I got 
[image]

Then, I ran `./ZED_Streaming_Receiver 192.168.0.20: 30000` on Ubuntu, and it still gets `[ZED][Streaming] Error: Stream opening timeout [Sample][Error] Camera Open | FAILURE : The function does not perform as expected. Exit program.`

obraun-sl, 2021-04-04:
Can you try on the same ubuntu PC? ie : 
./ZED_Streaming_sender
./ZED_Streaming_receiver localhost

to check first that it's working. 
On windows, once an application communicates on the network, you must make sure that comms are authorized for the app (you should have a popup window that ask if you want this app to access the network).

reginalluna, 2021-04-04:
> Can you try on the same ubuntu PC? ie :
> ./ZED_Streaming_sender
> ./ZED_Streaming_receiver localhost
>
> to check first that it's working.
> On windows, once an application communicates on the network, you [...]

Thank you so much for the detailed explanation.

Unfortunately, it doesn't work. 
When I ran the python scripts on PC (windows10), I got:
- sender: 
[image]
- receiver: 
[image]
---
When I ran the cpp scripts on PC (windows10), it ran successfully.
- sender:
[image]
- receiver:
[image]
 ----

I have a question think what the issue with the python scripts? How can I let it works? 

Another question, how can I run the sender and plug the ZED on Ubuntu18 (the laptop), and run the receiver on the PC (windows10)? I plan to get a Jetson NX later and carry it outside to capture some data, connect the Jetson device and the laptops (windows 10 or Ubuntu18) through the Wifi hotspot provided by my phone.

The last question, if I place Jetson with ZED outdoor, is it possible for Jetson to send the data to my PC placed from home (theoretically say, the Jetson device and PC will use different internet and let Jetson send the data remotely)?

I'm an artist, so I'm not really familiar with programming. If you can help with these question, I appreciate a lot!

closed 2022-04-26 by github-actions[bot]