stereolabs/zed-sdk #364: Camera streaming latency

jobesu14, 2021-04-11:
Hello,

First of all I would like to congratulate the Stereolabs dev team for the amazing effort your are putting on the crystal clear documentation and super useful examples. I have to say that I found myself enjoying equally playing with my new ZED Mini than exploring the code of the different repos. Congrats!

My use case:
I have a Jetbot robot running a Jetson Nano. I would like to get the robot position and show the camera feed on a remote computer. To do that I see two options:
1. send the camera stream through Wifi to the remote computer (on the local network) as it is done on the "Camera Streaming" example. In this case I would show the camera feed and do the position tracking of the robot on the remote computer.
2. Do the position tracking on the Jetson and send the position and the video feed to the remote computer.

The prefered option would be the 1. I therefore tried the "Camera Streaming" example with my ZED Mini both on my Windows 10 Laptop (i9 @ 2.4Ghz, RTX 2070 Max-Q) and my Jetson Nano.

With sender and receiver on same machine (localhost) I was expecting to have a latency in the range 20 - 50ms at least on my powerful Laptop, but the latency was more in the range of 500ms - 2s...
I did the same test on the Jetson Nano and I got pretty much the same latency value.

I tried to play with the stream `_params.bitrate` and `stream_params.chunk_size` parameters without any satisfying improvement.

I then tried to stream from the Jetson Nano (Wifi) to the Laptop (Ethernet) and here the latency was around 2-8s... Also the receiver often didn't succeed to connect to the sender and/or crashed.

Is those latency values what one should expect?
- If no, what latency value should we expect on:
        1. localhost
        2. local **Ethernet** network?
    What should I try to reduce the latency?

- If yes, what would be the best way to reduce the latency? Are you guys sending the frames in TCP? Is it possible to disable any buffering to lower the latency? Would it be better, latency wise to send the camera feed with zed-gstreamer? In this case, what latency min should I expect?

PS: in my experience, the latency when sending 1080p frames in local network over Wifi with a dummy python UDP client server is around 70ms - 150ms...

closed 2022-04-26 by github-actions[bot]

Rutherfordum, 2022-05-17:
Hello everyone, how can I increase the stream buffer in this script [script](https://github.com/stereolabs/zed-examples/blob/master/camera%20streaming/sender/python/streaming_sender.py)

obraun-sl, 2022-05-17:
Hi,
I am not sure what you mean by "stream buffer" but if it's the bitrate, you can modify it here : 
https://github.com/stereolabs/zed-examples/blob/master/camera%20streaming/sender/python/streaming_sender.py#L42

The bitrate must be set on KBits/s --> on the example it's 4Mb/s

Rutherfordum, 2022-05-18:
> Привет, я не уверен, что вы подразумеваете под «буфером потока», но если это битрейт, вы можете изменить его здесь: [https://github.com/stereolabs/zed-examples/blob/master/camera%20streaming/sender/ python/streaming_sender.py#L42](https://github.com/stereolabs/zed-examples/blob/master/camera%20streaming/sender/python/streaming_sender.py#L42)
> 
> Битрейт должен быть установлен в КБит/с --> в примере это 4Мбит/с

I want to increase the delay so that the frames are smooth, because. the UDP part is lost and I get micro-twitching

obraun-sl, 2022-05-19:
Ok. Try to use stream.chunk_size parameters and increase the value to 65000. That will reduce the number of chunk for each frames and that might help to reduce collision/lost of packet.