stereolabs/zed-sdk #369: [Question] Coordinate transform for Zed 2

Oreon94-TU, 2021-04-19:
I want to place the zed 2 camera in a large hall where I can get the skeleton data of the people working there. But since the camera is placed at a height and angled downwards, the coordinates I get are with the camera frame. How do I change the coordinates to a world frame with axes along the floor plane?

I was trying to do the quaternion transformation using the following code but gives no difference in result:

# Get IMU data
    zed.get_sensors_data(zed_sensors, sl.TIME_REFERENCE.IMAGE)
    zed_imu = zed_sensors.get_imu_data()
    zed_imu_pose = sl.Transform()
    zed_pose = sl.Pose()
    
    initial_translation = sl.Translation()
    initial_translation.init_vector(0,180,0)
    py_transform.set_translation(initial_translation)
    py_transform.set_orientation(zed_imu.get_pose(zed_imu_pose).get_orientation())
    positional_tracking_parameters.set_initial_world_transform(py_transform)

obraun-sl, 2021-04-20:
Hi,

You need to use RuntimeParameters::measure3D_reference_frame in world to output metrics in real world (otherwise they are in camera frame).
See here : https://www.stereolabs.com/docs/api/structsl_1_1RuntimeParameters.html#a95f5726cd4db955a263d913d310f38e4

Oreon94-TU, 2021-04-20:
> Hi,
> 
> You need to use RuntimeParameters::measure3D_reference_frame in world to output metrics in real world (otherwise they are in camera frame).
> See here : https://www.stereolabs.com/docs/api/structsl_1_1RuntimeParameters.html#a95f5726cd4db955a263d913d310f38e4

Does this use the frame at which the camera was set when it started?
For me, the camera is static and this did not seem to effect on the XYZ values.

closed 2022-04-26 by github-actions[bot]