stereolabs/zed-sdk #370: Imu raw data is wrong?

Xiaoyue-Wang, 2021-04-21:
When i use imu sensor raw data, I found imu_data.linear_acceeleration is near zero and imu_data.angular_velocity is near (0,0,-9.8)., which I believe it's data of linear_accleration. 
It's also the same result when I directly run code in tutorials-7 sensor data.
My sdk is window10-cuda11.1_v3.4.3.
do I understand something wrong ,Could you help me?

obraun-sl, 2021-04-21:
Can you specify the language your are using (C++, python, C#) ?
Also can you double check the SDK version? (3.4.3 does not exists).

Xiaoyue-Wang, 2021-04-21:
Oh, I'm sorry. I use the C++ and version is 3.4.2. code is like this.

void recordImuInfo(SensorsData::IMUData imu_data, std::ofstream& f,std::ofstream& p) {
    f <<"Translation : "<<imu_data.pose.getTranslation()<<",  Orientation:"<< imu_data.pose.getOrientation() << ", EulerAngles:"<<imu_data.pose.getEulerAngles()
      <<"," << imu_data.linear_acceleration 
      << "," << imu_data.angular_velocity 
      << ","<<imu_data.linear_acceleration_uncalibrated
      <<","<<imu_data.angular_velocity_uncalibrated
      <<","<<" Timestamp:"
        //cout << "\t Timestamp: {" << sensors_data.imu.timestamp.getMilliseconds() << " }\n ";
      //<< "," << zed.getTimestamp(TIME_REFERENCE::CURRENT).getMilliseconds() << " }\n ";
      << zed.getTimestamp(TIME_REFERENCE::CURRENT).data_ns << std::endl;
      
    p << zed.getTimestamp(TIME_REFERENCE::CURRENT).data_ns << ",   " << imu_data.pose.r00 << "," << imu_data.pose.r01 << "," << imu_data.pose.r02 << "," << imu_data.pose.tx << "," << imu_data.pose.r10 
           << "," << imu_data.pose.r11 << "," << imu_data.pose.r12 << "," << imu_data.pose.ty << "," << imu_data.pose.r20
           << "," << imu_data.pose.r21 << "," << imu_data.pose.r22 << "," << imu_data.pose.tz << "\n";

}

int main(int argc, char **argv) {

    
    // Set configuration parameters
    InitParameters init_params;
    init_params.camera_resolution = RESOLUTION::HD1080;
    init_params.depth_mode = DEPTH_MODE::ULTRA;
    init_params.coordinate_units = UNIT::METER;
    init_params.camera_fps = 30;
    

    if (argc > 1) init_params.input.setFromSVOFile(argv[1]);
        
    // Open the camera
    ERROR_CODE err = zed.open(init_params);
    if (err != ERROR_CODE::SUCCESS) {
        printf("%s\n", toString(err).c_str());
        zed.close();
        return 1; // Quit if an error occurred
    }

    // Display help in console
    printHelp();

    // Check camera model.
    auto info = zed.getCameraInformation();
    // Display sensors configuration (imu, barometer, magnetometer).

    printSensorConfiguration(info.sensors_configuration.accelerometer_parameters);
    printSensorConfiguration(info.sensors_configuration.gyroscope_parameters);
    printSensorConfiguration(info.sensors_configuration.magnetometer_parameters);
    printSensorConfiguration(info.sensors_configuration.barometer_parameters);
    
    sl::Transform trans = info.sensors_configuration.camera_imu_transform;
    std::cout <<"Imu_Camera Transform: {  "<< trans.getInfos().c_str() << " }"<<std::endl;
      
    
    // Used to store sensors data.
    SensorsData sensors_data;
    SensorsData::IMUData imu_data;

    std::thread threadCamera(getCameraData);
    threadCamera.detach();

    std::ofstream t;
    t.open("imuInfo.txt");
    char key = ' ';
    while (key != 'q') {
        if (zed.getSensorsData(sensors_data, TIME_REFERENCE::IMAGE) == ERROR_CODE::SUCCESS) {
            // Check if a new IMU sample is available. IMU is the sensor with the highest update frequency.
            imu_data = sensors_data.imu;

            
            /*
            std::cout << " - IMU:\n";
            std::cout << " \t Orientation: {" << sensors_data.imu.pose.getOrientation() << "}\n";
            std::cout << " \t Acceleration: {" << sensors_data.imu.linear_acceleration << "} [m/sec^2]\n";
            std::cout << " \t Angular Velocitiy: {" << sensors_data.imu.angular_velocity << "} [deg/sec]\n";
            //cout << "\t Timestamp: {" << sensors_data.imu.timestamp.getMilliseconds() << " }\n ";
            std::cout << "\t Timestamp: {" << zed.getTimestamp(TIME_REFERENCE::CURRENT).getMilliseconds() << " }\n ";
           */
            recordImuInfo(imu_data, t);
        }
    }

   
    
    zed.close();
    return 0;

Xiaoyue-Wang, 2021-04-21:
the result is like this:
Translation : 0.996502 -0.0579017 0,  Orientation:-0.49985 -0.50015 0.50015 0.49985, EulerAngles:0 -1.5708 1.5708,-0.523469 0.12207 0.244141,-0.123026 0.593725 -9.73381,-0.627281 1.96789e-06 0,-0.274658 0.490811 -9.69652, Timestamp:1618985441187958900
Translation : 0.996502 -0.0579017 0,  Orientation:-0.49985 -0.50015 0.50015 0.49985, EulerAngles:0 -1.5708 1.5708, linear_acc: -0.523469 0.12207 0.244141, angular_velocity: -0.123026 0.593725 -9.73381,linear_acc_uncalibration :-0.627281 1.96789e-06 0,angular_velocity_uncalib: -0.274658 0.490811 -9.69652, Timestamp:1618985441188564200

I use in win10. Also, could you help me to know what's the mearning of imu_data.pose.getTranslation()( result: 0.996502 -0.0579017 0)? (the value of 0.9965?) And I found when the camera is near static, the angular velocity or the linear_accleration(because the value I think is mixed)  has the value of 0.5?
the value in Zed Sensor Viewer.exe is right and is it the raw data in Zed Sensor Viewer.exe?

obraun-sl, 2021-04-21:
I cannot run your code because I am missing the getCameraData fct that runs into your detached thread.
Can you run the unmodified tutorial and show the output ?
ZED Sensor Viewer shows the calibrated data and the fused orientation. 
It's not a good idea to use the translation from the IMU as it's not precise since it's based on double integration of linear acceleration. 

It's better to use the positional tracking if you want to have access to the position of the camera

Xiaoyue-Wang, 2021-04-21:
I run the tutorial-7.  https://github.com/stereolabs/zed-examples/blob/master/tutorials/tutorial%207%20-%20sensor%20data/cpp/main.cpp

the result is like this:

Sample 11
- IMU:
    Orientation:{ -4.99775 -0.500225 -0.500621 0.499391}
    Acceleraion:{ 2.5613 -0.12207 0.335693}
  Angular Velocity: {-0.111366 -0.202105,-9.56061}

I couldm't understand the value of Acceleration and Angular Velocity.

Xiaoyue-Wang, 2022-04-20:
您好，您的邮件我已收到，我会尽快查收。祝好！

closed 2022-09-23 by Myzhar