stereolabs/zed-sdk #371: [?] Is Camera Streaming sample workable with rasberry pi setup?

FavoRiteKK, 2021-04-22:
Hi, I read the official https://www.stereolabs.com/docs/video/streaming/, and the hardware requirement describes this

> When streaming live video, hardware acceleration is used to perform real-time encoding and decoding with minimal overhead. It is available on Nvidia GeForce, Quadro, Tesla and embedded Jetson boards (Nano, TX2, Xavier).

I am confused, I wonder if I can install the sample sender on a rasberry pi 4 board, and the receiver on a nvidia machine? I mean, will my setup for camera streaming work?

Thanks!

obraun-sl, 2021-04-23:
Hi,
It does not work on RPI4 since the sender uses the NVIDIA hardware encoder for real-time H264/5 encoding for streaming.
For the Sender, you can use the Jetson Nano , which price is very close to RPI4B+

FavoRiteKK, 2021-04-23:
Thank you, @obraun-sl 
I'll  close this issue.

closed 2021-04-23 by FavoRiteKK