stereolabs/zed-sdk #372: ZED_Depth_Viewer failed to run

liuchang138929, 2021-04-23:
I have installed the CUDA11.0 and the "ZED_SDK_Ubuntu20_cuda11.0_v3.5.0.run"

CUDA version:
nvcc: NVIDIA (R) Cuda compiler driver
Copyright (c) 2005-2020 NVIDIA Corporation
Built on Thu_Jun_11_22:26:38_PDT_2020
Cuda compilation tools, release 11.0, V11.0.194
Build cuda_11.0_bu.TC445_37.28540450_0

But when I run the "ZED_Depth_Viewer" at /usr/local/zed/tools, it failed: 

[ZED Depth Viewer] Cuda Error: cudaErrorInvalidValue
[ZED Depth Viewer] Cuda Error: cudaErrorInvalidValue
Segmentation fault (core dumped)

obraun-sl, 2021-04-23:
Can you specify the graphic card you are using? 
Also the output of $ nvidia-smi

liuchang138929, 2021-04-23:
> Can you specify the graphic card you are using?
> Also the output of $ nvidia-smi

Thanks for your reply. 
The output of nvidia-smi is:

Fri Apr 23 21:25:17 2021       
+-----------------------------------------------------------------------------+
| NVIDIA-SMI 460.56       Driver Version: 460.56       CUDA Version: 11.2     |
|-------------------------------+----------------------+----------------------+
| GPU  Name        Persistence-M| Bus-Id        Disp.A | Volatile Uncorr. ECC |
| Fan  Temp  Perf  Pwr:Usage/Cap|         Memory-Usage | GPU-Util  Compute M. |
|                               |                      |               MIG M. |
|===============================+======================+======================|
|   0  GeForce RTX 207...  Off  | 00000000:01:00.0  On |                  N/A |
| 18%   33C    P8    25W / 215W |    164MiB /  7979MiB |      2%      Default |
|                               |                      |                  N/A |
+-------------------------------+----------------------+----------------------+
                                                                               
+-----------------------------------------------------------------------------+
| Processes:                                                                  |
|  GPU   GI   CI        PID   Type   Process name                  GPU Memory |
|        ID   ID                                                   Usage      |
|=============================================================================|
|    0   N/A  N/A      1164      G   /usr/lib/xorg/Xorg                112MiB |
|    0   N/A  N/A      1465      G   /usr/bin/gnome-shell               47MiB |
|    0   N/A  N/A      2522      G   /usr/lib/firefox/firefox            2MiB |
+-----------------------------------------------------------------------------+

apanagar, 2021-07-07:
It looks like you're running CUDA 11.2. I'm also trying to run with 11.2, but having some issues like 'GPU out of memory'. Does the SDK work with CUDA 11.2, or do I explicitly need to downgrade to 11.1?

[image]

Darker-zq, 2021-11-16:
There was an error when I ran ZED_Depth_Viewer on ubuntu18.04. It shown that "an NVIDIA GPU(Kepler or newer)is required to run this software. But I'm sure that I had been installed CUDA10.2 and Cudnn8.2. So what can i do for this?

obraun-sl, 2021-11-16:
Can you specify the graphic card you are using?
and can you provide also the output of $ nvidia-smi

Darker-zq, 2021-11-16:
[image]
my cuda_version is 10.2

obraun-sl, 2021-11-16:
I think the problem comes from an incompatibility between the cuda version and your driver version.
Try to uninstall CUDA10.2, Install CUDA11.4 and install the ZED SDK for cuda11.X version.

Darker-zq, 2021-11-16:
thanks your reply and i will try it.

Jeromeh97, 2022-02-02:
@Darker-zq Hi, have you tried the method @obraun-sl said? Let me know if you have fixed the issue. Cheers!

yucath, 2022-02-24:
Hi, were you able to solve the problem? I am having similar issue. I have ubuntu 20.04. using ZED SDK v3.6.5, cuda 11.5, nvidia rtx-3070 with driver version 510.47.03.

[image]

closed 2022-04-26 by github-actions[bot]

SoulProficiency, 2023-12-27:
> Hi, were you able to solve the problem? I am having similar issue. I have ubuntu 20.04. using ZED SDK v3.6.5, cuda 11.5, nvidia rtx-3070 with driver version 510.47.03.
> 
> [image]

have you solved this issue？