stereolabs/zed-sdk #380: Difference in distance value from DEPTH and POINT CLOUD!

chandailrc, 2021-05-12:
Hello,

I was looking to extract the distance values of certain objects in the frame. I followed the [depth-sensing example](https://www.stereolabs.com/docs/tutorials/depth-sensing/). The code I used is from the [zed-examples-tutorial 3](https://github.com/stereolabs/zed-examples/blob/master/tutorials/tutorial%203%20-%20depth%20sensing/python/depth_sensing.py).

The only additional line in the tutorial code that I have added is `err, depth_value = depth.get_value(x, y)`  above line 81, in order to compare the depth information obtained from MEASURE.DEPTH and MEASURE.XYZRGBA i.e. the point cloud. (I also added in a print statement to print the DEPTH based distance)

I expected the results to differ slightly, based on the tutorial, but what I see is a huge discrepancy between the distance obtained via. PointCloud (PC) vs DEPTH
...
PC Distance to Camera at (640, 360) (image center): 2.08 m
DEPTH Distance to Camera at (640, 360): 0.838 m
PC Distance to Camera at (640, 360) (image center): 2.06 m
DEPTH Distance to Camera at (640, 360): 0.836 m
PC Distance to Camera at (640, 360) (image center): 2.09 m
DEPTH Distance to Camera at (640, 360): 0.84 m
PC Distance to Camera at (640, 360) (image center): 2.08 m
DEPTH Distance to Camera at (640, 360): 0.839 m
...

The scene is static. The pixel location is static. As per the real world distance, the PC is giving the correct value. **However**, I have seen the reverse happen as well, where the DEPTH was giving correct values and PC was underestimating by a huge margin!

Am, I doing something wrong? I can post the code if needed, but I haven't changed anything significantly apart from the above mentioned changes.

qt-truong, 2021-05-17:
Hi @chandailrc,

You are right, the values for `sl.MEASURE.XYZ[RGBA]` and `sl.MEASURE.DEPTH` should be consistent for given pixel coordinates. There is an issue in the API and (x,y) are in fact reversed for your depth map. Thank you for reporting it, it will be fixed in the next release !

While this is being fixed, you can check that `depth.get_value(y,x)` is close to the distance calculated using your point cloud values. Note that your `PC Distance` is the **euclidian distance** of the object to the camera sensor while `DEPTH distance` is the distance of the object to the camera plane, so the two values can slightly differ.
However, your depth value should be equivalent to the Z value of your point cloud (`point_cloud_value[2]` in [tutorial 3](https://github.com/stereolabs/zed-examples/blob/21177250cfd96a735645dc12527fcb7daf41f5fa/tutorials/tutorial%203%20-%20depth%20sensing/python/depth_sensing.py#L76) )

Hope it helps !

chandailrc, 2021-05-21:
Hello qt-truong,

Thanks for looking into this! I did do some indoor tests and yes flipping to `depth.get_value(y,x)` makes the two Z values match i.e., the `point_cloud_value[2]` and the `depth.get_value()` for a specific (x,y) coordinate. 

I will be doing some long distance testing as well. That is where I had encountered the issue first and in the reverse order as well. It would be that PointCloud data wasn't able to estimate the correct distance but the depth.get_value seemed to come close to the actual distance. This correction in the coordinate order for `depth.get_value` would definitely change things.

Will report back if I see further discrepancies.

closed 2022-04-26 by github-actions[bot]

urbansound8K, 2022-09-27:
is it frixed???
why would you close it?

Myzhar, 2022-09-27:
Because it's fixed :)