stereolabs/zed-sdk #387: How could I get the keypoint 2D and 3D of persons?

Kismetzc, 2021-05-25:
I want to get the keypoint 2D and 3D coordinate of persons' 18 skeleton points in body  tracking/python/body_tracking.py
[https://github.com/stereolabs/zed-examples/blob/21177250cfd96a735645dc12527fcb7daf41f5fa/body%20tracking/python/body_tracking.py](url)

qt-truong, 2021-05-25:
Hi @Kismetzc,

To retrieve 3D keypoint coordinates, you can have a look at this part of the code : [body tracking sample L.579](https://github.com/stereolabs/zed-examples/blob/21177250cfd96a735645dc12527fcb7daf41f5fa/body%20tracking/python/ogl_viewer/viewer.py#L579) 
```python
for part in range(len(sl.BODY_PARTS)-1):
    kp = obj.keypoint[part]
```  
This loop allows you to retrieve the 3D keypoint coordinates of each [sl.BODY_PARTS](https://www.stereolabs.com/docs/api/python/classpyzed_1_1sl_1_1BODY__PARTS.html).
The current body_tracking sample only uses 3D keypoint coordinates for 3D display, but the API provides the attribute `keypoint_2d` to retrieve 2D keypoint coordinates. You would only need to replace `keypoint` by `keypoint_2d` in the loop above to retrieve them.

For information, you can also have a look at our old body tracking sample which uses 2D keypoint coordinates for 2D display with OpenCV : [body_tracking_2d.py](https://github.com/stereolabs/zed-examples/blob/ea9fdf306dc241a51e93db56dac2182dc8e3c1dc/body%20tracking/python/body_tracking_2d.py)

Kismetzc, 2021-05-26:
> Hi @Kismetzc,
> 
> To retrieve 3D keypoint coordinates, you can have a look at this part of the code : [body tracking sample L.579](https://github.com/stereolabs/zed-examples/blob/21177250cfd96a735645dc12527fcb7daf41f5fa/body%20tracking/python/ogl_viewer/viewer.py#L579)
> 
> ```python
> for part in range(len(sl.BODY_PARTS)-1):
>     kp = obj.keypoint[part]
> ```
> 
> This loop allows you to retrieve the 3D keypoint coordinates of each [sl.BODY_PARTS](https://www.stereolabs.com/docs/api/python/classpyzed_1_1sl_1_1BODY__PARTS.html).
> The current body_tracking sample only uses 3D keypoint coordinates for 3D display, but the API provides the attribute `keypoint_2d` to retrieve 2D keypoint coordinates. You would only need to replace `keypoint` with `keypoint_2d` in the loop above to retrieve them.
> 
> For information, you can also have a look at our old body tracking sample which uses 2D keypoint coordinates for 2D display with OpenCV : [body_tracking_2d.py](https://github.com/stereolabs/zed-examples/blob/ea9fdf306dc241a51e93db56dac2182dc8e3c1dc/body%20tracking/python/body_tracking_2d.py)

Thanks for your answer.  Actually, I want to get 3D points. Can I replace `keypoint` with `keypoint_3d` in the loop?

Kismetzc, 2021-05-26:
I know it. Does `keypoint`  mean 3D points?

Kismetzc, 2021-05-26:
By the way, why are the samples which are come with zed different from Github's samples? The `viewer.py` I saw in the samples attached to zed is different from the `viewer.py` you mentioned

qt-truong, 2021-05-26:
Hi @Kismetzc,

> Does `keypoint` mean 3D points?

Yes, `keypoint` attribute will output 3D point coordinates whereas `keypoint_2d` will give you 2D point coordinates.

> By the way, why are the samples which are come with zed different from Github's samples? The `viewer.py` I saw in the samples attached to zed is different from the `viewer.py` you mentioned

Sorry about that, the github repository has just been updated with the latest version of the samples (the ones that come with the SDK)

Kismetzc, 2021-05-26:
Hi @qt-truong,

I can get the 'keypoint' with your help! Thanks for your answer again!

I still have a question now. 

Is there any documentation about the coordinate origin?  I want to know where it is.

qt-truong, 2021-05-26:
If you don't specify your `RuntimeParameters` ([doc](https://www.stereolabs.com/docs/api/python/classpyzed_1_1sl_1_1RuntimeParameters.html)), your coordinates will be by default expressed in the Camera reference frame whose origin is at the center of the left sensor.
You can specify which coordinate system you want in `InitParameters.coordinate_system`, which can have the following values : [doc](https://www.stereolabs.com/docs/api/group__Core__group.html#ga1114207cbac18c1c0e4f77a6b36a8cb2). 
By default, `sl.COORDINATE_SYSTEM.IMAGE` will be used

Kismetzc, 2021-05-26:
OK. I can save these keypoints now!

Is the key point of my output for each frame?

Kismetzc, 2021-05-26:
> If you don't specify your `RuntimeParameters` ([doc](https://www.stereolabs.com/docs/api/python/classpyzed_1_1sl_1_1RuntimeParameters.html)), your coordinates will be by default expressed in the [...]

OK，let me read the documentation.

closed 2021-05-31 by Kismetzc