stereolabs/zed-sdk #388: Zed2 3d mapping

dhaval491, 2021-05-27:
Whenever I set 3d mapping flag to true on my zed2 camera, it stops the mapping node by giving the folllowing msg:
[ZED][Spatial mapping] [WARNING] Spatial mapping stopped: not enough memory

Any one know of how ths issue can be reolved. Do I need to put in extra sd card for this?

I am working on JetsonTx2 platformm

TheMadMoose, 2021-06-21:
Hey! I got this last week on our Jetson Nano as well. I'm not sure if I can be of any help, but it happened to me when I was trying to capture 2000 frames for generating a spatial map + texture. After the error, I did some research and all sorts of temporary solutions (for example...create a swap file, but apparently the SDK didn't know how to handle it or something? The reply from Stereolabs wasn't too helpful unfortunately, so I'm not sure where it's standing now...) came up and I didn't get the time to dig into it any further. I lowered the amount of frames to 1000 and set the maximum memory usage to 1500mb....It seemed to solve it for now. (I have the 2GB Jetson Nano).

Were you also trying to capture a texture along with the mesh? (That was my case at least)

I didn't get around to testing it any further yet (other matters came in the way), but my next step was to up the frames to 2000 again and test it once more....I was wondering if it's necessary to actually provide the spatial mapping method with maximum available memory to use....Could be that it will just flush it once it full when you do that, and otherwise will just clog the memory and we get the error?

Anyway, if I'll get any further with this I'll try to update here...

```
def spatialMapping(self, textureSaving=True, mapResolution=0.1):
        mapping_parameters = sl.SpatialMappingParameters()
        mapping_parameters.save_texture = textureSaving
        mapping_parameters.resolution_meter = mapResolution
        **mapping_parameters.max_memory_usage = 1500**
```

Or

P-yver, 2021-11-30:
Hi,
Indeed the ZED SDK incrementaly allocates memory when needed to stores the current mesh.
The memory (RAM) limit is a parameter; as @TheMadMoose  mentionned it: `mapping_parameters.max_memory_usage`.
On embedded platforms this limit is easly reached. You can either use a lower mesh resolution, or when you reach this limit, then extract the current map and save it and start again the mapping process,

closed 2022-04-26 by github-actions[bot]