stereolabs/zed-sdk #390: Can I save the result as an obj model in body tracking?

Kismetzc, 2021-05-31:
I ran the body tracking demo and saw skeleton in ZED Body Tracking view.  Can I save the result as an obj model?

Kismetzc, 2021-05-31:
Can I replace Simple3DObject class with my own obj model?  Achieve the effect that my obj model follows my movements.

Kismetzc, 2021-05-31:
Can I understand it this way? The 3D view is to connect each point of the bones for rendering.
It’s not about having a 3D model, letting the model track the movement.

I am very confused about this and hope to get your help

Kismetzc, 2021-06-02:
Hi,@qt-truong
I need your help! Can you take a look at this issue when you are free?

qt-truong, 2021-06-02:
Hello @Kismetzc, sorry for the late reply !

I am very sorry but I am not sure I understand what you are trying to do. Could you please elaborate on your problem ? 

Thanks in advance for your reply

Kismetzc, 2021-06-02:
Hello @qu-truong,
Thank you for answering right away.

Draw 3D bones in body_tracking.py, it can follow people's movements, now I want to replace the 3D bone model with a 3D human model. Let the 3D human model move with the recognition pose.

I hope I made my confusion clear.

Kismetzc, 2021-06-02:
[image]
This is body_tracking.py result.

I want a 3D human model to track my movements instead of the 3D skeleton model

Bvallon-sl, 2021-06-03:
Hi,

The body tracking sample only creates basic shapes (cylinders and sphere) in order to create a skeleton. It does not handle more complex applications like animating a rigged 3D Model.

I would recommend using our Unity3D plugin which include a example scene that reproduce exactly what you are looking for.
You can find more details about our Unity plugin [here](https://www.stereolabs.com/docs/unity/).

Best regards,
Benjamin Vallon

Kismetzc, 2021-06-04:
> Hi,
>
> The body tracking sample only creates basic shapes (cylinders and sphere) in order to create a skeleton. It does not handle more complex applications like animating a rigged 3D Model.
>
> I would [...]

Thanks for your answer! I will try!
By the way, can I achieve the purpose by overriding the `Simple3DObject` class?

Kismetzc, 2021-06-04:
[image]
To add, I want to achieve such an effect picture

obraun-sl, 2021-06-04:
This is exactly what the zed-unity plugin is offering. 
Doing that in C++ by overriding Simple3DObject class is much more complicated since you will basically need to recreate what's inside unity engine.

Kismetzc, 2021-06-05:
> This is exactly what the zed-unity plugin is offering.
> Doing that in C++ by overriding Simple3DObject class is much more complicated since you will basically need to recreate what's inside unity engine.

OK!

Kismetzc, 2021-06-06:
> This is exactly what the zed-unity plugin is offering.
> Doing that in C++ by overriding Simple3DObject class is much more complicated since you will basically need to recreate what's inside unity engine.

I ran the demo successfully! Thanks for your help!

Kismetzc, 2021-06-06:
Hi, @qt-truong 
And I have one more question.
I am still a beginner in OpenGL. Is the skeleton of the 3D perspective obtained by rendering the 2D perspective?

closed 2021-08-08 by Kismetzc