stereolabs/zed-sdk #393: Fail to open ZED depth viewer on Win10

SimonSongg, 2021-06-13:
I installed the latest SDK 3.5.0 and I cannot run ZED depth viewer. No error, just nothing happened after I run the application.
Win 10, CUDA 10.2, RTX2080super

SimonSongg, 2021-06-13:
I tried the ZED diagnostic, everything was correct.

Myzhar, 2021-06-13:
@SimonSongg can you please write an email to support@stereolabs.com adding the report generated by the ZED Diagnostic tool as an attachment?

1187938150, 2021-07-21:
When I disconnect the wifi connection, I can open the depth viewer. Otherwise I can't open it.

Jeromeh97, 2022-02-02:
@SimonSongg Have you sorted this issue? I have the same problem as you had.

Myzhar, 2022-02-02:
If you are using an antivirus try to disable it.
Some kind of antivirus stop the ZED SDK if not correctly configured.

closed 2022-04-26 by github-actions[bot]