stereolabs/zed-sdk #394: Terminal ZED Explorer command for recording outputs not working properly

Sohob, 2021-06-17:
Hello,

I'm trying the Body Tracking python sample, but whenever I try recording with the ZED Explorer command through terminal e.g, "./ZED_Explorer -r VGA -o vga.svo" the output svo cannot be processed. It gets stuck at "Using SVO file: /path/to/svo" and opens a black window.

What's even weirder is that recording directly from the ZED Explorer application gets processed without a problem. Also the svo's are played fine using the ZED Explorer without even a display issue.

I tried tracing the problem, and so far I found that the zed.grab() always returns END_OF_SVOFILE_REACHED. T

Many Thanks!

obraun-sl, 2021-06-18:
Hello,
Are you on Windows or Linux based system ?

Sohob, 2021-06-18:
Linux, Ubuntu 20.04

Sohob, 2021-06-22:
Tried on windows 10 also, same problem

closed 2022-04-26 by github-actions[bot]