stereolabs/zed-sdk #395: The init_parameters.camera_fps does not work in the Tutorial 2 - image capture

robNavLoc, 2021-06-25:
**### When I set the parameter to 10 fps or 30 fps, the outputs shown almost the same timestamps.** Why?

This is the code of the Tutorial 2

`#include <sl/Camera.hpp>

using namespace std;
using namespace sl;

int main(int argc, char **argv) {

    // Create a ZED camera object
    Camera zed;

    // Set configuration parameters
    InitParameters init_parameters;
    init_parameters.camera_resolution = RESOLUTION::HD1080; // Use HD1080 video mode
    init_parameters.camera_fps = 30; // Set fps at 30

    // Open the camera
    auto returned_state = zed.open(init_parameters);
    if (returned_state != ERROR_CODE::SUCCESS) {
        cout << "Error " << returned_state << ", exit program." << endl;
        return EXIT_FAILURE;
    }

    // Capture 50 frames and stop
    int i = 0;
    sl::Mat image;
    while (i < 50) {
        // Grab an image
        returned_state = zed.grab();
        // A new image is available if grab() returns ERROR_CODE::SUCCESS
        if (returned_state == ERROR_CODE::SUCCESS) {

            // Get the left image
            zed.retrieveImage(image, VIEW::LEFT);
            
            // Display the image resolution and its acquisition timestamp
            cout<<"Image resolution: "<< image.getWidth()<<"x"<<image.getHeight() <<" || Image timestamp: "<<image.timestamp.data_ns<<endl;
            i++;
        }
    }

    // Close the camera
    zed.close();
    return EXIT_SUCCESS;
}`
**And the result shows that 50 images speet 3.717s in 30fps and 3.867s in 10fps**

So, what's wrong with it?

Myzhar, 2021-06-26:
Hi @robNavLoc 
what is the configuration of your host device?

robNavLoc, 2021-06-26:
Hi， Thanks for your reply.
My host device is :
CPU: Inter Core ***@***.***GHz
GPU: Nvidia Geforce 940MX

BUT！ When I set the fps to 10, it should spend 5 seconds for loading 50 images, but it spent 3.7 seconds. 
So I think it may have few influence by my device. Right? And I want to know the reson of the results.

Thanks very very very much!!!

-----------------------------------

Myzhar, 2021-06-26:
10 FPS is not a valid setting. It is automatically modified to 15 FPS:
https://www.stereolabs.com/docs/video/camera-controls/
The fact that you are not getting more than 10 FPS means that your host device has not enough computational power to run at full speed.
Can you verify the performances reported by the `ZED Depth Viewer` tool?

robNavLoc, 2021-06-26:
> 10 FPS is not a valid setting. It is automatically modified to 15 FPS:
> https://www.stereolabs.com/docs/video/camera-controls/
> The fact that you are not getting more than 10 FPS means that your host [...]

It shows that 
[ZED Depth Viewer] Cuda Error: cudaErrorInvalidGraphicsContext
[ZED Depth Viewer] Cuda Error: cudaErrorInvalidValue
I don't know why. The Cuda in my ubuntu18.04 is cuda11.0 

But the other tools can be run.

So it just because the host device has not enough computational power?

But the other tools like ZED_Explorer shows that the Framerate is 60 fps or 30 fps.

closed 2022-04-26 by github-actions[bot]