stereolabs/zed-sdk #398: Not enough GPU Memory error

apanagar, 2021-07-07:
I Installed the ZED Sdk. I tried running ZED_Explorer, this starts running and I can see the camera output. Strangely, it is bright and then becomes dim, I'm not sure if this is normal. I then close that and run ZED_Depth_Viewer and get the following error:

[image]

Screenshot of ZED_Diagnostic tool (json below):
[image]

Here is an image of the 'dimmed' camera views
[image]

Not exactly sure what to do next, should I remove CUDA 11.2? Should I completley uninstall CUDA and reinstall 11.1 clean?

Below is my diagnostic output from ZED_Diagnostic_Results. Note that I am running CUDA 11.2, but also installed 11.1. Based on the below output, I'm assuming 11.1 is what is being used??:

{
    "Camera Test": {
        "ZEDCount": 1,
        "deviceInitResult": 0,
        "firmware": 1523,
        "resolutions": {
            "HD1080@15": {
                "fps (input)": 15,
                "fps (output)": 15,
                "height (input)": 1080,
                "height (output)": 1080,
                "initCode": true,
                "initialization": "Ok",
                "width (input)": 1920,
                "width (output)": 1920
            },
            "HD1080@30": {
                "fps (input)": 30,
                "fps (output)": 30,
                "height (input)": 1080,
                "height (output)": 1080,
                "initCode": true,
                "initialization": "Ok",
                "width (input)": 1920,
                "width (output)": 1920
            },
            "HD2K@15": {
                "fps (input)": 15,
                "fps (output)": 15,
                "height (input)": 1242,
                "height (output)": 1242,
                "initCode": true,
                "initialization": "Ok",
                "width (input)": 2208,
                "width (output)": 2208
            },
            "HD720@15": {
                "fps (input)": 15,
                "fps (output)": 15,
                "height (input)": 720,
                "height (output)": 720,
                "initCode": true,
                "initialization": "Ok",
                "width (input)": 1280,
                "width (output)": 1280
            },
            "HD720@30": {
                "fps (input)": 30,
                "fps (output)": 30,
                "height (input)": 720,
                "height (output)": 720,
                "initCode": true,
                "initialization": "Ok",
                "width (input)": 1280,
                "width (output)": 1280
            },
            "HD720@60": {
                "dirty lenses": 0,
                "dropCount": -1,
                "fps (input)": 60,
                "fps (output)": 60,
                "height (input)": 720,
                "height (output)": 720,
                "imageCount": 502,
                "initCode": true,
                "initialization": "Ok",
                "tearingCount": 0,
                "width (input)": 1280,
                "width (output)": 1280
            },
            "VGA@100": {
                "fps (input)": 100,
                "fps (output)": 100,
                "height (input)": 376,
                "height (output)": 376,
                "initCode": true,
                "initialization": "Ok",
                "width (input)": 672,
                "width (output)": 672
            },
            "VGA@15": {
                "fps (input)": 15,
                "fps (output)": 15,
                "height (input)": 376,
                "height (output)": 376,
                "initCode": true,
                "initialization": "Ok",
                "width (input)": 672,
                "width (output)": 672
            },
            "VGA@30": {
                "fps (input)": 30,
                "fps (output)": 30,
                "height (input)": 376,
                "height (output)": 376,
                "initCode": true,
                "initialization": "Ok",
                "width (input)": 672,
                "width (output)": 672
            },
            "VGA@60": {
                "fps (input)": 60,
                "fps (output)": 60,
                "height (input)": 376,
                "height (output)": 376,
                "initCode": true,
                "initialization": "Ok",
                "width (input)": 672,
                "width (output)": 672
            }
        },
        "serial": 17381639,
        "valid": [
            "<b>Camera: </b>ZED Mini",
            "<b>Firmware: </b>1523"
        ],
        "videoNumber": 0
    },
    "Graphics Card": {
        "deviceCount": 1,
        "deviceDriverVersion": 11020,
        "devices": [
            {
                "arch": "Pascal",
                "computeCapability": "6.1",
                "cores": 2560,
                "name": "GeForce GTX 1080",
                "totalMemoryBytes": "8512602112",
                "totalMemoryMB": 8118.25
            }
        ],
        "initResult": 0,
        "valid": [
            "<b>Graphics card: </b> GeForce GTX 1080"
        ]
    },
    "Processor": {
        "OS": "Ubuntu 20.04.2 LTS ",
        "avxSupported": true,
        "coreCount": "12",
        "cpu": " AMD Ryzen 5 1600 Six-Core Processor",
        "datetime": "2021-07-07 10:20:48 AM",
        "hyperThreading": "1",
        "motherboard": "X370 Killer SLI/ac, ASRock",
        "ramCapacity": "16.785912",
        "ramUsage": "",
        "ramUsed": "2.810540",
        "threadCount": "24",
        "valid": [
            "<b>Processor: </b>  AMD Ryzen 5 1600 Six-Core Processor",
            "<b>Motherboard: </b> X370 Killer SLI/ac, ASRock"
        ]
    },
    "USB": {
        "CorruptedFirmware": false,
        "MCUDetected": true,
        "USBList": [
            {
                "USBMode": 3,
                "USB_path": "/1",
                "bDescriptorType": 1,
                "bDeviceProtocol": 1,
                "bLength": 18,
                "bMaxPacketSize0": 9,
                "bNumConfigurations": 1,
                "bcdDevice": "1.0",
                "bcdUSB": "3.0",
                "bcdUSBClass": 239,
                "bcdUSBSubClass": 2,
                "busNumber": 4,
                "device": "ZED mini",
                "iManufacturer": 1,
                "iProduct": 2,
                "iSerial": 0,
                "idProduct": "0xf682",
                "idVendor": "0x2b03"
            },
            {
                "USBMode": 2,
                "USB_path": "/1",
                "bDescriptorType": 1,
                "bDeviceProtocol": 0,
                "bLength": 18,
                "bMaxPacketSize0": 64,
                "bNumConfigurations": 1,
                "bcdDevice": "2.5",
                "bcdUSB": "2.0",
                "bcdUSBClass": 0,
                "bcdUSBSubClass": 0,
                "busNumber": 3,
                "device": "ZED mini MCU",
                "iManufacturer": 1,
                "iProduct": 2,
                "iSerial": 3,
                "idProduct": "0xf681",
                "idVendor": "0x2b03"
            },
            {
                "USB_path": "/9",
                "idProduct": "0x0aa7",
                "idVendor": "0x8087"
            },
            {
                "USB_path": "/6",
                "idProduct": "0x8176",
                "idVendor": "0x0bda"
            }
        ],
        "USBMode": 3,
        "ZED Camera Module Detected": 63106,
        "ZED MCU Module Detected": 63105,
        "ZEDDetected": true,
        "valid": [
            "<b>USB Bandwidth: </b> OK"
        ]
    },
    "ZED SDK": {
        "CUDA Toolkit version": "V11.1.105",
        "DiagnosticZEDSDKVersion": "3.5.0",
        "InstalledZEDSDKVersion": "Unknown",
        "SMI": "(SMI) :     Performance State                     : P0",
        "initResult": 1,
        "loaded": true,
        "valid": [
            "<b>CUDA version:</b> 11.1"
        ],
        "warning": [
            "Multiple version of CUDA has been detected. It is recommended to uninstall other versions than 11.1."
        ]
    }
}

apanagar, 2021-07-19:
This issue was due to using a VNC with the zed Tools.

closed 2021-07-19 by apanagar

jianjuan, 2021-08-13:
Hello,  I met the same problems, can you tell me how to solve this ?

apanagar, 2021-08-13:
I was connected via vnc remotely to a headless server with a gpu. I solved this by connecting my monitor to the video card. In my case the opengl context of the zed explorer was routing gl calls to the video card display rather than the vnc virtual desktop . Virtual desktops aren't rendered by the physical video card so that is a problem.

When I figured out the issue, I used x11vnc and a hardware dongle to trick the video card into thinking a display was connected. x11vnc allows streaming the physical desktop of the machine. 

For clarity, the error was caused by opengl trying to allocate memory somewhere other than the videocard.

jianjuan, 2021-08-13:
Hi, I do not understand this clearly, can I make a conclusion that my problem is caused by dual system(window10 and ubuntu20.04)? Even I know that "the error was caused by opengl trying to allocate memory somewhere other than the videocard", I still have no idea how to fix it, please help, thank you very much !

obraun-sl, 2021-08-13:
No it's not coming from a dual system. 
As @apanagar said, the error probably comes from the fact that the graphics memory (opengl) is not allocated on the same video card as the ZED SDK is running on (NVIDIA). 
Make sure that your display is connected to the NVIDIA GPU because OpenGL automatically creates its context on the video card the display is connected on. 

For ex, one common mistake is to plug the screen to the DVI port of the motherboard instead of the the hdmi/dvi port of the nvidia graphic card

jianjuan, 2021-08-13:
I think I have connected right:
[image]
but it still doesn't work, I fell so confuesd, crying....
[image]

can you figure out  other problems?

obraun-sl, 2021-08-13:
Can you post the diagnostic file from ZED Diagnostic tool ?

jianjuan, 2021-08-13:
Okay, I just run the ZED_Diagnostic tools, and the diagnostic file shows:

jianjuan, 2021-08-13:
I have solved this problem! When I install nvidia driver, I added  "-no-opengl-files", reinstall nvidia driver without this parameter will be nice! 
thank you guys very much!