stereolabs/zed-sdk #399: Lacking example: saving left/right image plus depth map and IMU data at the same time once a second

arasharchor, 2021-07-08:
Hi, I am new to ZED. 

I have a ZED camera and want to install it on a car. I intend to save left/right image plus depth map and IMU data at the same time once a second, but I could not find a sample for that in Python3.

Is there any sample which I can use for this purpose to jump in?

qt-truong, 2021-07-09:
Hi @arasharchor,

- You can have a look at our [depth sensing tutorial](https://github.com/stereolabs/zed-examples/blob/master/tutorials/tutorial%203%20-%20depth%20sensing/python/depth_sensing.py) that will show you how to retrieve images and depth from your camera

- The [sensors data tutorial](https://github.com/stereolabs/zed-examples/blob/master/tutorials/tutorial%207%20-%20sensor%20data/python/sensor_data.py) will show you how to access your ZED IMU data. 

Don't hesitate to visit the [API Reference](https://www.stereolabs.com/docs/api/python/) and modify the tutorials according to your needs

luan1412167, 2021-07-16:
@arasharchor @qt-truong do you have any script for getting images, depth, IMU together?

closed 2022-04-26 by github-actions[bot]