stereolabs/zed-sdk #404: Is an IMU orientation calculated from gyroscope and accelerometer only?

yyc9268, 2021-07-21:
Hi,
I feel curious about how the IMU orientation is caculated.
I refered this [code](https://github.com/stereolabs/zed-examples/blob/master/tutorials/tutorial%207%20-%20sensor%20data/python/sensor_data.py).
In the code, 'sensors_data.get_imu_data().get_pose().get_orientation().get()' function returns quaternion orientation.
Since its an child function of 'get_imu_data()' I guess its been filtered orientation fusing only IMU sensor data (gyroscope and accelerometer).
In my best knowledge, accelerometer can prevent drifts of roll and pitch of gyroscope data. 
The orientation seems very accurate than I have expected.
Could any staff in Stereolabs give me the official comments about this?
I wonder whether its been calculated from only IMU or been aided by additional sensors like magnetometer or barometer.

Thanks,

obraun-sl, 2021-07-21:
Hi,
Orientation of sensors_data is calculated by using gyroscope and accelerometer only. Currently magnetometer is not used in the imu fusion. Accelerometer is used to compensate drifts for pitch and roll. 
For yaw, only integration of the gyroscope can be used. 
Note that we also have internally a factory calibration of the IMU (that will adjust bias of gyro and accelero) + a runtime calibration of the bias of the gyroscope. This means that when the camera is static, the bias of the gyroscope will be adjusted to compensate drift along yaw axis.

yyc9268, 2021-07-21:
@obraun-sl 
Thank you very much. 
It is awesome that the pose from IMU only fusion is accurate enough.

closed 2021-07-21 by yyc9268