stereolabs/zed-sdk #409: How can I include the debug version of ZED SDK libraries?

Rajesh-VR, 2021-08-11:
Hi All,

I am trying to create a test project which is from https://www.stereolabs.com/blog/how-to-set-up-a-multi-camera-network/ But when I include "sl/Camera.hpp (#include <sl/Camera.hpp>)", I am getting the below build (Debug build) errors:

    /usr/local/zed/include/sl/Camera.hpp:95: error: #warning ("'Debug' builds are not supported since this library was built in 'Release', 'RelWithDebInfo' should be preferred to avoid crashes and memory issues") [-Werror=cpp]

So how can I include the debug version of ZED SDK libraries?

Regards,
Rajesh

obraun-sl, 2021-08-11:
This should be a warning, not an error. 
In any case, try to call cmake with -DCMAKE_BUILD_TYPE=Release argument to see if it solves

Rajesh-VR, 2021-08-12:
Hi @obraun-sl,
Thanks for your reply. The release build works without any issue, but I wish to make the debug build for the same.
If I define "ALLOW_BUILD_DEBUG", then I can take the debug build. Is there any other option ?
Regards

closed 2022-04-26 by github-actions[bot]