stereolabs/zed-sdk #411: Want to know sl::Mat ref-counted or not?

Rajesh-VR, 2021-08-16:
Hi All,

I am creating the sample project from  https://github.com/stereolabs/zed-examples/tree/master/tutorials/tutorial%202%20-%20image%20capture/cpp

I wish to know that sl::Mat is ref-counted or not?

Thanks in advance.

Regards,
Rajesh

adujardin, 2021-08-16:
Unfortunately, it's not. 

You can check if the Mat is a weak reference or not using [`sl::Mat::isMemoryOwner()`](https://www.stereolabs.com/docs/api/classsl_1_1Mat.html#a277ae510e34fb6dbb561efe4fbe35934) or [`sl::Mat::getInfos()`](https://www.stereolabs.com/docs/api/classsl_1_1Mat.html#ab49a456b67d2135d93334a2b1737d247). The sl::Mat memory owner should therefore not be freed to avoid any unwanted behavior when using reference Mat (`sl::Mat::move` or `sl::Mat::clone` can be used instead).

Rajesh-VR, 2021-08-17:
Thanks @adujardin for the valuable response, it is more helpful.

closed 2021-08-17 by Rajesh-VR