stereolabs/zed-sdk #414: Issue with tutorial 4: "Error CAMERA MOTION SENSORS NOT DETECTED, exit program."

jiegusc, 2021-08-25:
My device is ZED 2i. I am running the program on Ubuntu 18.04.
I think this device should support the body track function.
However when I run the tutorial 4, the output showed ‘CAMERA MOTION SENSORS NOT DETECTED’.
Could anyone tell me how to solve this problem? 

Thank you

jiegusc, 2021-08-25:
@ymtoo

ymtoo, 2021-08-25:
[Tutorial 4](https://github.com/stereolabs/zed-examples/blob/master/tutorials/tutorial%204%20-%20positional%20tracking/python/positional_tracking.py) works fine on my setup (ZED 2i, Ubuntu 18.04, ZED SDK 3.5.2, pyzed 3.5). Can you run ZED_Diagnostic and other tools in /usr/local/zed/tools?

obraun-sl, 2021-08-25:
@jiegusc Can you post the result of lsusb when camera is connected?

jiegusc, 2021-08-25:
> @jiegusc Can you post the result of lsusb when camera is connected?

jiegusc, 2021-08-25:
> @jiegusc Can you post the result of lsusb when camera is connected?

Thank you for your response. Here is the result of lsusb on my computer.

Bus 002 Device 004: ID 2b03:f880  
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 003: ID 17ef:608d Lenovo 
Bus 001 Device 002: ID 17ef:6099 Lenovo 
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

[image]

jiegusc, 2021-08-25:
> [Tutorial 4](https://github.com/stereolabs/zed-examples/blob/master/tutorials/tutorial%204%20-%20positional%20tracking/python/positional_tracking.py) works fine on my setup (ZED 2i, Ubuntu 18.04, ZED SDK 3.5.2, pyzed 3.5). Can you run ZED_Diagnostic and other tools in /usr/local/zed/tools?

The ZED_Diagnostic works fine.

However I don't know how to use ZED_Sensor_viewer. Could you please give me some hint？ 

Thank you

jiegusc, 2021-08-25:
> @jiegusc Can you post the result of lsusb when camera is connected?

And when I run the tutorial 6 and 8, The [ZED][INFO] showed "Sensors not detected". Please check the image below.
[image]

obraun-sl, 2021-08-25:
@jiegusc Contact the support at support@stereolabs.com 
It seems your camera is defective on the MCU side as it does not appears on the device list. 
When writting to the support, copy the link of this github thread in your mail. Thanks.

jiegusc, 2021-08-25:
> @jiegusc Contact the support at [support@stereolabs.com](mailto:support@stereolabs.com)
> It seems your camera is defective on the MCU side as it does not appears on the device list.
> When writting to the support, copy the link of this github thread in your mail. Thanks.

Got it.

Thank you for helping out.

closed 2021-08-25 by jiegusc