stereolabs/zed-sdk #420: Recording on the Jetson Nano

yuvalg1987, 2021-08-30:
Hi, 

I'm trying to record a SVO file on the Jetson Nano but it crashes (both the ZED explorer and the code example).  I'm getting the following error :
```
[NvV4l2Elements] failed to open
Error in VIDIOC_REQBUFS at output planeError in VIDIOC_REQBUFS at output plane[ZED][Encoding] Critical Error : No Video Enc
Segmentation fault
```

I'm using the following release of Jetpack and the latest ZED SDK
`# R32 (release), REVISION: 6.1, GCID: 27863751, BOARD: t210ref, EABI: aarch64,
`

This error has been previously reported but still no solution. Any advise?

Thanks
Yuval

obraun-sl, 2021-08-30:
It seems that you have installed Jetpack4.6. Note that the ZED SDK is not yet compatible with it. 
I suggest you flash the JP4.5.1 instead. 

Make sure that the tegra multimedia API is selected during flash because it's the one that install the needed components to access HW video encoder ( ` [ZED][Encoding] Critical Error : No Video Enc ` means that no video encoder has been found)

yuvalg1987, 2021-08-30:
Thanks for the quick reply! How can I check that the Multimedia API is properly installed? I can already see that Gstremer is properly installed (I assumed that it includes the Multimedia API). 

I haven't used Jetpack for the flash, just downloaded the updated image for the Nvidia site. It will be strange if they haven't included it in their default image. 

it seems that this issue has been around for some time now [https://github.com/stereolabs/zed-python-api/issues/107](url)

yuvalg1987, 2021-08-30:
Ok I verified that the Multimedia API is properly installed using the following command:
`sudo apt list | grep nvidia.*multimedia*`

I assume that the only way to check that it is properly configured is to run a gstreamer app that uses the encoders./decoders by Nvidia

yuvalg1987, 2021-08-30:
I've tested the the Nvidia decoder using the following gst coomand and it works, I still want to test some encoding functions
` gst-launch-1.0 filesrc location=vid_example.mp4 ! qtdemux ! queue ! h264parse ! nvv4l2decoder ! nv3dsink -e`

Any ideas what more should I test?

yuvalg1987, 2021-08-30:
Ok, found a solution. The issue is caused by the installation of libv4l, following the instruction in here 
[solved](https://github.com/stereolabs/zed-gstreamer/issues/19#issuecomment-756617653
) it.

closed 2021-08-30 by yuvalg1987

Jade999, 2022-01-27:
@yuvalg1987 thank you for your sharing , this solved my problem.