stereolabs/zed-sdk #422: Save pointcloud from SVO file  for each frame

Pavol89, 2021-09-05:
The app ZEDfu allows creating a fuzed pointcloud. Is it possible to export pointclouds for each frame from the SVO file?

In ZED depth viewer it is possible to save a pointcloud for one frame selection, but it cannot be done for the whole svo file continuously.

Thanks for any advice.

obraun-sl, 2021-09-07:
You can use the fonction sl::Mat::write() with a filename ending with ".ply" for example and increment the filename with a frame counter.

Pavol89, 2021-09-07:
Thanks for response. I used comand in depth sensing in 72th rows in original code [:https://github.com/stereolabs/zed-examples/blob/master/depth%20sensing/cpp/src/main.cpp]

```
// Allocation of 4 channels of float on GPU
    Mat point_cloud(camera_config.resolution, MAT_TYPE::F32_C4, MEM::GPU);
int svo_position = zed.getSVOPosition();
sl::Mat::write(point_cloud,("pointcloud_" + to_string(svo_position) + ".ply").c_str());
```

But I get this error:
```
/home/pclinux/catkin_ws/src/zed-examples-master/depth sensing/cpp/src/main.cpp:73:86: error: no matching function for call to ‘sl::Mat::write(sl::Mat&, const char*)’
 :write(point_cloud,("pointcloud_" + to_string(svo_position) + ".ply").c_str());
```

obraun-sl, 2021-09-07:
write() is a member function of sl::Mat (not a static fct). This is the correct syntax : 

Code : 
```
          int svo_position = zed.getSVOPosition();
          point_cloud.write(("pointcloud_" + to_string(svo_position) + ".ply").c_str());
```

Pavol89, 2021-09-09:
Thanks for help.

closed 2022-04-26 by github-actions[bot]