stereolabs/zed-sdk #424: what are the minimum dependancies for zed image capture?[Question]

Rajesh-VR, 2021-09-06:
Hi All,

I am trying to create the sample project which will capture the images using zed2i camera ([sample code from sterolabs](https://github.com/stereolabs/zed-examples/tree/master/camera%20streaming/sender)). For this purpose I just created new lxc and copied "libsl_zed.so" into lxc. When I run "lsusb" on lxc, then it is listing the zed2i, but I am getting the result of "sl::Camera::getDeviceList()" as ZERO.

So I wish to know the minimum dependencies for running the image capturing methods for ZED. Kindly help.
Note: I don't want to install the whole zed sdk into my lxc.

Thanks,
Rajesh

Myzhar, 2021-09-06:
If you do not want to use the ZED SDK, you can try the [ZED Open Capture](https://github.com/stereolabs/zed-open-capture)

Myzhar, 2021-09-08:
This seems an error related to OpenCV:
https://github.com/opencv/opencv/issues/18461

Rajesh-VR, 2021-09-10:
now the "hid_enumerate(SL_USB_VENDOR, 0x0);" call returns NULL (zed_oc_sensors_example) even in direct machine and also the same in lxc.

closed 2022-04-26 by github-actions[bot]