stereolabs/zed-sdk #426: Segmentation Fault while optimising model

njenwei, 2021-09-09:
Hi

I am encountering a segmentation fault in ZED_Object_detection_birds_eye_viewer when the model is being optimised for my PC.

My PC's configuration is Ryzen 9, RTX3070, Ubuntu 20.04, Nvidia driver 470, CUDA 11.0, ZED SDK 3.5.3.

I have tried reinstalling the SDK and CUDA several times.

This is the output I get, which is the same for all the other samples which require the AI model.
```
$ ./ZED_Object_detection_birds_eye_viewer 
[ZED][Init] Depth mode: ULTRA
[ZED][Init] Video mode: HD1080@30
[Sample] Object Detection: Loading Module... 

[ZED][Object Detection] Please wait while the AI model is being optimized for your graphics card
 This operation will be run only once and may take a few minutes  \Segmentation fault (core dumped)
```

This is a gdb output:
```
[Sample] Object Detection: Loading Module... 

[ZED][Object Detection] Please wait while the AI model is being optimized for your graphics card
 This operation will be run only once and may take a few minutes [New Thread 0x7fff17fed700 (LWP 28954)]
 \--Type <RET> for more, q to quit, c to continue without paging--

Thread 1 "ZED_Object_dete" received signal SIGSEGV, Segmentation fault.
0x00007fff6c0ff9d8 in APIToModel(unsigned int, ai::YOLOV5_MODEL_VARIANT, unsigned long, unsigned long, nvinfer1::IHostMemory**, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, unsigned long, bool, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >) () from /usr/local/zed/lib/libsl_ai.so
(gdb) backtrace
#0  0x00007fff6c0ff9d8 in APIToModel(unsigned int, ai::YOLOV5_MODEL_VARIANT, unsigned long, unsigned long, nvinfer1::IHostMemory**, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, unsigned long, bool, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >) () at /usr/local/zed/lib/libsl_ai.so
#1  0x00007fff6c0ffe87 in TensorrtxExecutor::createEngine() () at /usr/local/zed/lib/libsl_ai.so
#2  0x00007fff6c1019a7 in TensorrtxExecutor::init(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, int, CUstream_st*, CUctx_st*&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, int, unsigned long) () at /usr/local/zed/lib/libsl_ai.so
#3  0x00007fff6c0ee318 in OD_h::initNet(int, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, int, int, CUstream_st*, CUctx_st*&, unsigned long) () at /usr/local/zed/lib/libsl_ai.so
#4  0x00007ffff0cac6aa in ai::ObjectDetector::init(int, int, sl::Resolution, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, bool, CUctx_st*, CUstream_st*, ai::OD_FILTER) () at /usr/local/zed/lib/libsl_zed.so
#5  0x00007ffff0e7d8ae in sl::ObjectsDetectorHandler::init(sl::Resolution, int, int, sl::ObjectsDetectorHandler::MODE, CUctx_st*&, PerformanceMeasurer*, ai::OD_FILTER, float, float) ()
    at /usr/local/zed/lib/libsl_zed.so
#6  0x00007ffff0e6c027 in sl::Camera::enableObjectDetection(sl::ObjectDetectionParameters) () at /usr/local/zed/lib/libsl_zed.so
#7  0x000055555556672b in main(int, char**) (argc=1, argv=0x7fffffffdc78) at /usr/local/zed/samples/object detection/birds eye viewer/cpp/src/main.cpp:107
```

Is there a fix for this, or can I disable model optimisation?

Thanks

adujardin, 2021-09-09:
Hi

Did you install a custom version of TensorRT and/or cudnn (that might be different from the one provided with the ZED SDK)?

Could you run `ldd /usr/local/zed/lib/libsl_ai.so` and also `ls /usr/local/cuda*` ?

Could you also try to purge any model by removing `/usr/local/zed/resources/*` and `/tmp/.x6*` and retry?

The optimization process is a required step to run the AI model, it creates an inference engine optimized for your hardware (and lib versions), and it's handled by TensorRT. The segfault at this step typically implies mismatch versions of libraries.

njenwei, 2021-09-09:
Hi,

CUDA 11.0 and the other libraries were installed from Nvidia's repo.

I've also tried removing `/usr/local/zed/resources/*` and `/tmp/.x6*` based on other issues that I've seen here.

```
$ ldd /usr/local/zed/lib/libsl_ai.so
        linux-vdso.so.1 (0x00007fffa6094000)
        libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007f2514538000)
        libdl.so.2 => /lib/x86_64-linux-gnu/libdl.so.2 (0x00007f2514532000)
        librt.so.1 => /lib/x86_64-linux-gnu/librt.so.1 (0x00007f2514527000)
        libpng16.so.16 => /usr/lib/x86_64-linux-gnu/libpng16.so.16 (0x00007f25144ef000)
        libz.so.1 => /lib/x86_64-linux-gnu/libz.so.1 (0x00007f25144d3000)
        libjpeg.so.8 => /usr/lib/x86_64-linux-gnu/libjpeg.so.8 (0x00007f251444e000)
        libcuda.so.1 => /usr/lib/x86_64-linux-gnu/libcuda.so.1 (0x00007f2512b40000)
        libnvinfer.so.7 => /usr/local/cuda-11.0/lib64/libnvinfer.so.7 (0x00007f24f9c43000)
        libnvinfer_plugin.so.7 => /usr/local/cuda-11.0/lib64/libnvinfer_plugin.so.7 (0x00007f24f91b1000)
        libnvonnxparser.so.7 => /usr/local/cuda-11.0/lib64/libnvonnxparser.so.7 (0x00007f24f8d40000)
        libarchive.so.13 => /usr/lib/x86_64-linux-gnu/libarchive.so.13 (0x00007f24f8c7e000)
        libsl_zed.so => /usr/local/zed/lib/libsl_zed.so (0x00007f24f0d53000)
        libstdc++.so.6 => /usr/lib/x86_64-linux-gnu/libstdc++.so.6 (0x00007f24f0b6f000)
        libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007f24f0a20000)
        libgomp.so.1 => /usr/lib/x86_64-linux-gnu/libgomp.so.1 (0x00007f24f09de000)
        libgcc_s.so.1 => /lib/x86_64-linux-gnu/libgcc_s.so.1 (0x00007f24f09c3000)
        libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007f24f07d1000)
        /lib64/ld-linux-x86-64.so.2 (0x00007f2514d35000)
        libcublas.so.11 => /usr/local/cuda-11.0/lib64/libcublas.so.11 (0x00007f24ea981000)
        libcudnn.so.8 => /usr/local/cuda-11.0/lib64/libcudnn.so.8 (0x00007f24ea750000)
        libmyelin.so.1 => /usr/local/cuda-11.0/lib64/libmyelin.so.1 (0x00007f24e9f29000)
        libnvrtc.so.11.0 => /usr/local/cuda-11.0/lib64/libnvrtc.so.11.0 (0x00007f24e873e000)
        libnettle.so.7 => /usr/lib/x86_64-linux-gnu/libnettle.so.7 (0x00007f24e8704000)
        libacl.so.1 => /usr/lib/x86_64-linux-gnu/libacl.so.1 (0x00007f24e86f9000)
        liblzma.so.5 => /lib/x86_64-linux-gnu/liblzma.so.5 (0x00007f24e86ce000)
        libzstd.so.1 => /usr/lib/x86_64-linux-gnu/libzstd.so.1 (0x00007f24e8625000)
        liblz4.so.1 => /usr/lib/x86_64-linux-gnu/liblz4.so.1 (0x00007f24e8604000)
        libbz2.so.1.0 => /lib/x86_64-linux-gnu/libbz2.so.1.0 (0x00007f24e85f1000)
        libxml2.so.2 => /usr/lib/x86_64-linux-gnu/libxml2.so.2 (0x00007f24e8437000)
        libturbojpeg.so.0 => /usr/lib/x86_64-linux-gnu/libturbojpeg.so.0 (0x00007f24e8398000)
        libusb-1.0.so.0 => /lib/x86_64-linux-gnu/libusb-1.0.so.0 (0x00007f24e837c000)
        libnvcuvid.so.1 => /usr/lib/x86_64-linux-gnu/libnvcuvid.so.1 (0x00007f24e7c44000)
        libnvidia-encode.so.1 => /usr/lib/x86_64-linux-gnu/libnvidia-encode.so.1 (0x00007f24e7a27000)
        libcublasLt.so.11 => /usr/local/cuda-11.0/lib64/libcublasLt.so.11 (0x00007f24dc898000)
        libicuuc.so.66 => /usr/lib/x86_64-linux-gnu/libicuuc.so.66 (0x00007f24dc6b2000)
        libudev.so.1 => /lib/x86_64-linux-gnu/libudev.so.1 (0x00007f24dc685000)
        libicudata.so.66 => /usr/lib/x86_64-linux-gnu/libicudata.so.66 (0x00007f24dabc2000)

$ ls /usr/local/cuda*
/usr/local/cuda:
bin  doc  EULA.txt  extras  include  lib64  libnvvp  nsightee_plugins  nvml  nvvm  samples  Sanitizer  share  src  targets  tools  version.txt

/usr/local/cuda-11:
bin  compute-sanitizer  doc  DOCS  EULA.txt  extras  gds  include  lib64  libnvvp  nsightee_plugins  nvml  nvvm  README  samples  share  src  targets  tools  version.json

/usr/local/cuda-11.0:
bin  doc  EULA.txt  extras  include  lib64  libnvvp  nsightee_plugins  nvml  nvvm  samples  Sanitizer  share  src  targets  tools  version.txt

/usr/local/cuda-11.2:
bin  compute-sanitizer  DOCS  EULA.txt  extras  include  lib64  libnvvp  nsightee_plugins  nvml  nvvm  nvvm-prev  README  samples  share  src  targets  tools  version.json

/usr/local/cuda-11.4:
bin  compute-sanitizer  doc  DOCS  EULA.txt  extras  gds  include  lib64  libnvvp  nsightee_plugins  nvml  nvvm  README  samples  share  src  targets  tools  version.json

$ cat /usr/local/cuda/version.txt
CUDA Version 11.0.228
```

adujardin, 2021-09-09:
Thank you,

By "other libraries were installed from Nvidia's repo" did you mean CUDA or also TensorRT/cudnn?

Even if you have multiple CUDA versions, it seems to correctly take the libraries from CUDA 11.0.

Could you try other models such as [human body](https://www.stereolabs.com/docs/api/group__Object__group.html#ga4a212b90726adaaf0fae801b1cc913f4)?
Do you have any issues running other samples?

Just in case could you post the output of `nvidia-smi` command please?

njenwei, 2021-09-09:
The libraries cuda, libcudnn8-dev and nvidia-driver-470.

Yes, I have multiple versions of cuda to support tensorflow, etc, but the system path is configured for CUDA 11.0.

Other models including body tracking sample, and the ros-wrapper node also segfaults on that step.

```
$ nvidia-smi
Thu Sep  9 22:32:18 2021
+-----------------------------------------------------------------------------+
| NVIDIA-SMI 470.63.01    Driver Version: 470.63.01    CUDA Version: 11.4     |
|-------------------------------+----------------------+----------------------+
| GPU  Name        Persistence-M| Bus-Id        Disp.A | Volatile Uncorr. ECC |
| Fan  Temp  Perf  Pwr:Usage/Cap|         Memory-Usage | GPU-Util  Compute M. |
|                               |                      |               MIG M. |
|===============================+======================+======================|
|   0  NVIDIA GeForce ...  Off  | 00000000:08:00.0  On |                  N/A |
|  0%   45C    P8    23W / 270W |    127MiB /  7979MiB |      0%      Default |
|                               |                      |                  N/A |
+-------------------------------+----------------------+----------------------+

+-----------------------------------------------------------------------------+
| Processes:                                                                  |
|  GPU   GI   CI        PID   Type   Process name                  GPU Memory |
|        ID   ID                                                   Usage      |
|=============================================================================|
|    0   N/A  N/A      1429      G   /usr/lib/xorg/Xorg                115MiB |
|    0   N/A  N/A      1683      G   /usr/bin/gnome-shell               10MiB |
+-----------------------------------------------------------------------------+

```

adujardin, 2021-09-09:
Thank you, 

This version of the SDK is expecting cudnn 8.0.1 could you confirm that the version you are using is newer or equal (it should be something like `/usr/local/cuda-11.0/lib64/libcudnn_cnn_infer.so.8.0.1`)?

When running the ros wrapper did you enabled any AI model? Do other samples without OD module, like depth-sensing run without issues?

njenwei, 2021-09-10:
Hi

Yes I have no issues running the ros wrapper with depth, pos tracking and mapping. It only seg faults when `od_enabled` is true for all models (0 - 3).

```
$ ls /usr/local/cuda-11.0/lib64/ | grep cudnn
libcudnn_adv_infer.so
libcudnn_adv_infer.so.8
libcudnn_adv_infer.so.8.0.1
libcudnn_cnn_infer.so
libcudnn_cnn_infer.so.8
libcudnn_cnn_infer.so.8.0.1
libcudnn_ops_infer.so
libcudnn_ops_infer.so.8
libcudnn_ops_infer.so.8.0.1
libcudnn.so
libcudnn.so.8
libcudnn.so.8.0.1
```

adujardin, 2021-09-10:
Everything seems to be in order, that's quite surprising to still have this crash. I'm currently using an almost identical setup both hardware and software-wise (without issues obviously).

Did you upgrade the ZED SDK version? Was it working on other versions maybe?

Could you test running a sample in docker using the ZED SDK image?

- [To setup nvidia docker you can follow this tutorial](https://www.stereolabs.com/docs/docker/install-guide-linux/)
- Then you can run :

```bash
docker run --gpus all -it --privileged stereolabs/zed:3.5-devel-cuda11.0-ubuntu20.04
# Then download, compile and run the OD tutorial (need a ZED2 plugged in)
apt update ; apt install libpng-dev -y
wget https://github.com/stereolabs/zed-examples/archive/refs/heads/master.zip
unzip master.zip
cd "zed-examples-master/tutorials/tutorial 6 - object detection/cpp"
mkdir build; cd build;cmake ..;make
./ZED_Tutorial_6
```

Void12987, 2022-03-04:
Note for anyone else having this issue, I just waited 28 minutes and the ZED worked after that. A segmentation fault might be because there's not enough RAM or swap space allocated. Not sure...

closed 2022-04-26 by github-actions[bot]