stereolabs/zed-sdk #429: Coredump generated from ZED_SVO_Recording

felrock, 2021-09-21:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, feature request, or anything other than a bug report directly related to this project.

### Description

Running ZED_SVO_Recording on docker image stereolabs/zed:3.5-py-devel-jetson-jp4.5 on a Jetson AGX Device generates a coredump where pthread_mutex_lock.c is missing.

### Steps to Reproduce

1. Build ZED_SVO_Recording(inside docker container)
2. Run ./ZED_SVO_Recording test.svo

### Expected Result

Recording of video.

### Actual Result

Core dump.

### ZED Camera model

ZED2

### Environment

```shell
Linux for Tegra
Jetson Xavier AGX
Docker image: stereolabs/zed:3.5-py-devel-jetson-jp4.5
```

adujardin, 2021-09-21:
Hi, 
Did you run the image without `--privileged`? 
This can happen when trying to open the camera IMU without root privileged. In that case, you can set up the udev rules using [this script on the **host**](https://gist.github.com/adujardin/2d5ce8f000fc6a7bd40bee2709749ff8)

felrock, 2021-09-21:
Hi, yes.  I Ran this command `sudo docker run --runtime nvidia -e NVIDIA_DRIVER_CAPABILITIES=video,compute,utility --network="host" --privileged -it -v `pwd`:/src/workspace -w /src/workspace stereolabs/zed:3.5-py-devel-jetson-jp4.5 /bin/bash`

closed 2022-04-26 by github-actions[bot]