stereolabs/zed-sdk #443: Change the resolution of SVO files

kankanzheli, 2021-11-09:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, bug report, or anything other than a feature request directly related to this project.

### Proposal

I'm sorry. I am a novice. So there is such a problem. I wonder if there is any existing tool to change the resolution of the image in the SVO file? I obtained these SVO files in 1080p format. But I want to obtain RGB and depth images of 720p later. Even point clouds are generated with 720p data. What existing tools can do this? If so, how to use it? If not, can we add tools in this regard?

Myzhar, 2021-11-09:
Hi @kankanzheli 
the resolution of an SVO file cannot be changed, but you can use them at a lower resolution by retrieving images and depth data with a rescaling factor.
See `retrieveImage` and `retrieveMeasure` and the parameter `image_size`:
* `retrieveImage`: https://www.stereolabs.com/docs/api/classsl_1_1Camera.html#a01dce4f0af6f8959a9c974ffaca656b5
* `retrieveMeasure`: https://www.stereolabs.com/docs/api/classsl_1_1Camera.html#a9e0773c0c14ce5156c1fa2fde501c13e

kankanzheli, 2021-11-09:
I prefer python. Can I modify the output resolution by modifying "svo_export. py"? How should I modify it if I can?

Myzhar, 2021-11-09:
Sure, Python and C++ API functions are similar:
* retrieveImage: https://www.stereolabs.com/docs/api/python/classpyzed_1_1sl_1_1Camera.html#a6c03c5c98bc9e63fe09ec6a29297c6d8
* retrieveMeasure: https://www.stereolabs.com/docs/api/python/classpyzed_1_1sl_1_1Camera.html#ae387f1b1c5a1454f2e2812c77267cf60

kankanzheli, 2021-11-09:
In the svo_export.py, the resolution of data is defined as 
    # Get image size
    image_size = zed.get_camera_information().camera_resolution
    width = image_size.width
    height = image_size.height
    width_sbs = width * 2

How can I change it to output the 720p data?
svo_export.py is located in /usr/local/zed/samples/svo recording/export/python

Myzhar, 2021-11-09:
You cannot modify the SVO, you must use the `image_size` parameter when your application is running with the SVO as input.
When you call `retrieveImage` and `retrieveMeasure` after a `grab` command you set the resolution to the values that you need.

`svo_export.py ` is an example showing how to save live camera data in SVO format, not something to change the resolution of an SVO.

kankanzheli, 2021-11-09:
You just replied to me. That is, I can't modify the SVO file. Only the output of SVO can be modified. So my last question is how to get 720p image results. The [svo_export. Py] example is used to output SVO files as RGB and depth maps. And avi format video.

So is it not feasible to modify the image output of 720p on [svo_export. Py]? Do I have to recreate a python file to get the output image I want?

kankanzheli, 2021-11-09:
所以我不懂应该如何修改。是不是在这段代码中添加呢？
if app_type == AppType.LEFT_AND_RIGHT:
                zed.retrieve_image(right_image, sl.VIEW.RIGHT)
添加Resolution(1280, 720) ? Like this?
            if app_type == AppType.LEFT_AND_RIGHT:
                zed.retrieve_image(right_image, sl.VIEW.RIGHT, sl.Resolution(1280, 720))
            elif app_type == AppType.LEFT_AND_DEPTH:
                zed.retrieve_image(right_image, sl.VIEW.DEPTH, sl.Resolution(1280, 720))
            elif app_type == AppType.LEFT_AND_DEPTH_16:
                zed.retrieve_measure(depth_image, sl.MEASURE.DEPTH, sl.Resolution(1280, 720))

Myzhar, 2021-11-09:
You can look at the image capture example:
https://github.com/stereolabs/zed-examples/tree/master/tutorials/tutorial%202%20-%20image%20capture/python

This is the line that you must change:
https://github.com/stereolabs/zed-examples/tree/master/tutorials/tutorial%202%20-%20image%20capture/python
simply by adding the `Resolution(1280,720)` parameter:
`zed.retrieve_image(image, sl.VIEW.LEFT),resolution = Resolution(1280,720)`

kankanzheli, 2021-11-09:
I did those changes:
            # Retrieve SVO images
            zed.retrieve_image(left_image, sl.VIEW.LEFT),resolution = Resolution(1280,720)

            if app_type == AppType.LEFT_AND_RIGHT:
                zed.retrieve_image(right_image, sl.VIEW.RIGHT),resolution = Resolution(1280,720)
            elif app_type == AppType.LEFT_AND_DEPTH:
                zed.retrieve_image(right_image, sl.VIEW.DEPTH),resolution = Resolution(1280,720)
            elif app_type == AppType.LEFT_AND_DEPTH_16:
                zed.retrieve_measure(depth_image, sl.MEASURE.DEPTH),resolution = Resolution(1280,720)
But there is a error:
  File "svo_export.py", line 150
    zed.retrieve_measure(depth_image, sl.MEASURE.DEPTH),resolution = Resolution(1280,720)
    ^
SyntaxError: can't assign to function call

Myzhar, 2021-11-09:
Sorry, please remove `resolution = ` and replace it with `sl.`

kankanzheli, 2021-11-09:
I just do those changes:
            # Retrieve SVO images
            zed.retrieve_image(left_image, sl.VIEW.LEFT),sl.Resolution(1280,720)

            if app_type == AppType.LEFT_AND_RIGHT:
                zed.retrieve_image(right_image, sl.VIEW.RIGHT),sl.Resolution(1280,720)
            elif app_type == AppType.LEFT_AND_DEPTH:
                zed.retrieve_image(right_image, sl.VIEW.DEPTH),sl.Resolution(1280,720)
            elif app_type == AppType.LEFT_AND_DEPTH_16:
                zed.retrieve_measure(depth_image, sl.MEASURE.DEPTH),sl.Resolution(1280,720)

But the resolution of the output image is not changed. The resolution still is (1920,1080). Is there anything else that needs to be modified?

Myzhar, 2021-11-09:
What is the example that you are modifying?

kankanzheli, 2021-11-09:
located in /usr/local/zed/samples/svo recording/export/python/svo_export. py. Which is used for transform the svo data to image or avi video.

qt-truong, 2021-11-09:
Hi @kankanzheli,

There is a small typo in your code which triggers the `SyntaxError` message (the resolution parameter is outside the brackets). But as @Myzhar said, specifiying the desired resolution in `retrieve_image` and `retrieve_measure` should do it. You need to change `svo_export.py` sample from [L.141](https://github.com/stereolabs/zed-examples/blob/b0ad7a91de4b64dbe3296a5bb5ae802ec1913a20/svo%20recording/export/python/svo_export.py#L141) which should now become :

```python
# Retrieve SVO images
zed.retrieve_image(left_image, sl.VIEW.LEFT, resolution = sl.Resolution(1280,720))

if app_type == AppType.LEFT_AND_RIGHT:
    zed.retrieve_image(right_image, sl.VIEW.RIGHT, resolution = sl.Resolution(1280,720))
elif app_type == AppType.LEFT_AND_DEPTH:
    zed.retrieve_image(right_image, sl.VIEW.DEPTH, resolution = sl.Resolution(1280,720))
elif app_type == AppType.LEFT_AND_DEPTH_16:
    zed.retrieve_measure(depth_image, sl.MEASURE.DEPTH, resolution = sl.Resolution(1280,720))
``` 

Can you try it out and tell us if it works ? Thanks

kankanzheli, 2021-11-09:
OK! Now it is worked!

closed 2021-11-09 by qt-truong

rtarun1, 2025-11-07:
How to set frame rate? @Myzhar