stereolabs/zed-sdk #444: Manual filtering using texture confidence

Andrei-Lapusteanu, 2021-11-13:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, bug report, or anything other than a feature request directly related to this project.

### Proposal

Hello! I am not entirely sure if what I propose can already be achieved, but I could not code this in / find an answer, and as I do not think this is a bug, I am writing this in the form of a feature request.

I am trying to manually filter data (such as depth or point cloud data) using the **Texture Confidence** data. 

I am aware that I can filter the data using `sl::RuntimeParameters::confidence_threshold` and `sl::RuntimeParameters::texture_confidence_threshold`, but these filter out points which do not meet the thresholds, which I do not want.

I am also aware I can retrieve the **Confidence Measurement** using the `retrieveMeasure` function - but this only gives me access to filter the data using the **Confidence** data. I cannot find a way to set both the `confidence_threshold` and `texture_confidence_threshold` to 100 and then manually filter using the **Texture Confidence**.

If this is not currently possible, I would like to propose this feature request.

### Use-Case

This would allow to **manually** filter the data using the **Texture Confidence** (as of now, manual filtering can be achieved using only the **Confidence**).

P-yver, 2021-12-02:
Hi,
Changing the value of `sl::RuntimeParameters::confidence_threshold` will internally use the data of `MEASURE::CONFIDENCE` to filter the disparity/depth in output.
Currently the confidence Mat related to `sl::RuntimeParameters::texture_confidence_threshold` can not be retrieved, is this what you want?

Andrei-Lapusteanu, 2021-12-05:
Hi @P-yver, thank you for the response!

Yes, this is the information I would have liked to retrieve (related to `sl::RuntimeParameters::texture_confidence_threshold`). I have managed to force this by doing 2 grab calls per frame. In the first call I retrieve an image filtered using the texture confidence, and in the second grab, in which I use the points filtered in the first step, I do the rest of my processing. But this is a viable solution only if I run my application offline (using an .svo recording), it's not suitable for real-time acquisition.

closed 2022-01-22 by Andrei-Lapusteanu

Nunzio03, 2025-03-12:
I Think this Issue should be opened again, as having a retrievable texture confidence map would be useful in certain real-time applications.
In my case I'm dynamically filtering the point cloud in order to get just a sub-set of points with a higher confidence, but i do not want to filter the point-cloud a priori by giving a texture_confidence_treshold, i want to be able to choose how to filter based on the texture confidence map.

Is that just used internally? and if it is so, why is it not returnable to the user?