stereolabs/zed-sdk #446: Zed camera streaming for ipv6

felrock, 2021-11-18:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, bug report, or anything other than a feature request directly related to this project.

### Proposal

It would be nice for camera streaming to support ipv6, if I enter an ipv6 address into receiver/cpp  it prompts me with "Bad IP address specified"

### Use-Case

Use zed streaming over ipv6

### Anything else?

You guys rock

Myzhar, 2021-11-18:
Hi @felrock 
thank you for the request. We noted and it will be considered for the next releases of the ZED SDK.

closed 2023-04-11 by felrock

s-m-asjad, 2024-05-21:
was this resolved? Any documentation to refer to?