stereolabs/zed-sdk #448: Much faster to `zed.grab()` on demand rather than continually in a loop thread.

XDGFX, 2021-11-26:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, feature request, or anything other than a bug report directly related to this project.

### Description

Running the `zed.grab()` method in a separate thread to a callback to fetch images is considerably slower and less consistent than running `zed.grab()` immediately at the start of every callback.

### Steps to Reproduce

Using the Python API, the two options are described below:

1. Have a separate thread running `self.zed.grab()` inside a class. Callbacks which 'get' an image can then call `self.zed.retrieve_image(...)` as soon as a request is received (e.g. it could be a web request).
2. Have `self.zed.grab()` at the start of every callback, before then calling `self.zed.retrieve_image(...)` to fetch the requested image.

### Expected Result

Option 1 uses more CPU as the program is always doing something, even when there are no requests for images. But requests are faster because when a callback is triggered, the most recent image can be retrieved instead of having to `grab` a brand new one.

### Actual Result

Option 1 does use more CPU, however, unexpectedly, each image request takes considerably longer, and is much less consistent.

_Example from first method:_
Request 1:
```log
   ncalls  tottime  percall  cumtime  percall filename:lineno(function)
        1    0.066    0.066    0.066    0.066 {method 'retrieve_image' of 'pyzed.sl.Camera' objects}
        1    0.002    0.002    0.002    0.002 {method 'get_data' of 'pyzed.sl.Mat' objects}
```

Request 2:
```log
   ncalls  tottime  percall  cumtime  percall filename:lineno(function)
        1    0.067    0.067    0.067    0.067 {method 'get_data' of 'pyzed.sl.Mat' objects}
        1    0.064    0.064    0.065    0.065 {method 'retrieve_image' of 'pyzed.sl.Camera' objects}
```

Request 3:
```log
   ncalls  tottime  percall  cumtime  percall filename:lineno(function)
        1    0.198    0.198    0.198    0.198 {method 'retrieve_image' of 'pyzed.sl.Camera' objects}
        1    0.065    0.065    0.066    0.066 {method 'get_data' of 'pyzed.sl.Mat' objects}
```

_Example from second method:_
Request 1:
```log
   ncalls  tottime  percall  cumtime  percall filename:lineno(function)
        1    0.035    0.035    0.035    0.035 {method 'retrieve_image' of 'pyzed.sl.Camera' objects}
        1    0.015    0.015    0.015    0.015 {method 'grab' of 'pyzed.sl.Camera' objects}
        1    0.002    0.002    0.002    0.002 {method 'get_data' of 'pyzed.sl.Mat' objects}
```

Request 2:
```log
   ncalls  tottime  percall  cumtime  percall filename:lineno(function)
        1    0.037    0.037    0.037    0.037 {method 'retrieve_image' of 'pyzed.sl.Camera' objects}
        1    0.009    0.009    0.009    0.009 {method 'grab' of 'pyzed.sl.Camera' objects}
        1    0.001    0.001    0.001    0.001 {method 'get_data' of 'pyzed.sl.Mat' objects}
```

Request 3:
```log
   ncalls  tottime  percall  cumtime  percall filename:lineno(function)
        1    0.038    0.038    0.038    0.038 {method 'retrieve_image' of 'pyzed.sl.Camera' objects}
        1    0.006    0.006    0.006    0.006 {method 'grab' of 'pyzed.sl.Camera' objects}
        1    0.002    0.002    0.002    0.002 {method 'get_data' of 'pyzed.sl.Mat' objects}
```

### ZED Camera model

ZED2

### Environment

```shell
- Jetson Xavier NX
- Ubuntu 18.04.5 (Running in Docker from `nvcr.io/nvidia/l4t-base:r32.6.1`)
- CUDA confirmed working with `deviceQuery` sample inside container
- ZED SDK v3.6.0
```

XDGFX, 2021-11-26:
Also for reference, option 3 (which seems like a terrible idea) is to have `zed.grab()` in _both_ a separate thread and at the start of each callback. This option is slower than either of the other options.

P-yver, 2021-12-02:
Hi,
The [grab function](https://www.stereolabs.com/docs/api/classsl_1_1Camera.html#afa3678a18dd574e162977e97d7cbf67b) does not only retrieve images, it also performs computation.
It is not recommended to do grab and retrieve Image in different threads as the ZED SDK is meant to do them serially.
You can do this in a threads and then share the retrieved data with your own app thread.

closed 2022-04-26 by github-actions[bot]