stereolabs/zed-sdk #449: Is the rgb image returned by the pyzed API with distortion?

JimXu1989, 2021-11-27:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, bug report, or anything other than a feature request directly related to this project.

### Proposal

I want to make sure wether the rgb image return by pyzed have distortion or not?
If has distortion, how to diminish it?
And is the 3d point map returned by pyzed API align to the image with distortion?

### Use-Case

I want to calculate the radius of a round wood board accuratly, due to the 3d points of the edge is not stable. 
I calculate the fitted plane of the wood board first, then I project the pixel coordinate of the wood board edge back to the fitted plane. While, the radius value I got always bigger than real radius.
The same thing happen when I calculate the distance of two points, which all lay on the board.

Thanks

qt-truong, 2021-11-30:
Hi @JimXu1989,

When retrieving your image via the API, you can choose whether to retrieve the `sl.VIEW.LEFT_UNRECTIFIED` or `sl.VIEW.RECTIFIED` (here is the corresponding doc : [`sl.VIEW`](https://www.stereolabs.com/docs/api/python/classpyzed_1_1sl_1_1VIEW.html) and [`retrieve_image`](https://www.stereolabs.com/docs/api/python/classpyzed_1_1sl_1_1Camera.html#a6c03c5c98bc9e63fe09ec6a29297c6d8))
If you don't want "distortions" in your image, you will need to retrieve the `sl.VIEW.LEFT`

JimXu1989, 2021-12-01:
> Hi @JimXu1989,
>
> When retrieving your image via the API, you can choose whether to retrieve the `sl.VIEW.LEFT_UNRECTIFIED` or `sl.VIEW.RECTIFIED` (here is the corresponding doc : [...]

Thanks, so the  example provide by stereolab, which use sl.VIEW.LEFT, already return left images without distortions, am I right?

qt-truong, 2021-12-01:
@JimXu1989 yes exactly

closed 2024-11-25 by mattrouss