stereolabs/zed-sdk #452: My ZED SDK doesn't support CUDA 10.2, Ubuntu

pipidx, 2021-12-15:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, feature request, or anything other than a bug report directly related to this project.

### Description

After installation ZED SDK on my Ubuntu several tries with different latest versions. 
I could only run /usr/local/zed/tools/ZED_Explorer and it works fine. But rather than this version I couldn't run. Ex: ZED_Depth_Viewer.

Note that, I'm using CUDA 10.2  and [NVIDIA-Driver 495.44 ]

### Steps to Reproduce

1. When I ran /usr/local/zed/tools/ZED_Explorer the Zed camera worked well, I can adjust resolution as well.
2. When I ran /usr/local/zed/tools/ZED_Depth_Viewer, it didn't work as it said "NO GPU COMPATIBLE : An NVIDIA GPU (Kepler or newer) is required to run this software" 
3. Also when I tried to build some sample code and ran it became like this=> "[Sample][Error] Open Camera  | NO GPU COMPATIBLE : An NVIDIA GPU (Kepler or newer) is required to run this software." 
...

### Expected Result

I expected it to display and work like /usr/local/zed/tools/ZED_Explorer

### Actual Result

But the actual result I got is "NO GPU COMPATIBLE : An NVIDIA GPU (Kepler or newer) is required to run this software"

### ZED Camera model

ZED2i

### Environment

```shell
OS: Ubuntu 18.04.5 LTS
CPU: 11th Gen Intel® Core™ i7-11700K @ 3.60GHz × 16 
GPU: NVIDIA-Driver 495.44 
ZED SDK Version: e.g. v3.5.6
CUDA: 10.2
```

### Anything else?

About that problem, I used to reinstalled CUDA 11.1 and downloaded ZED SDK version accordingly and it worked.
By when I try my ZED2i with CUDA 10.2, it just didn't work even during the installation CUDA 10.2 could be detected.

P-yver, 2021-12-15:
Hi,
what GPU does your computer have? 
you can run the following command to know it:
> nvidia-smi -L

pipidx, 2021-12-15:
As I ran 

nvidia-smi -L

P-yver, 2021-12-16:
Support for 3070 (8.6 arch) is added by [Cuda 11.1](https://docs.nvidia.com/cuda/archive/11.1.0/cuda-toolkit-release-notes/index.html#cuda-general-new-features) : _Added support for NVIDIA Ampere GPU architecture based GA10x GPUs GPUs (compute capability 8.6), including the GeForce RTX-30 series._

pipidx, 2021-12-16:
I'm sorry, I couldn't get it. Does it mean I have to use Cuda 11.1?

P-yver, 2021-12-16:
yes, at least Cuda 11.1, I recommend to use the latest one supported by the ZED SDK (11.5)
ps: cuda minor versions are now compatible.

pipidx, 2021-12-16:
Ah okay, I will work on it then. Thank you very much for your answer.

closed 2021-12-16 by P-yver