stereolabs/zed-sdk #453: GPU ID for unity

AymanMukh, 2021-12-21:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, bug report, or anything other than a feature request directly related to this project.

### Proposal

I have a pc with multiple GPUs, when I use unity, it always uses GPU number 0. Is there a way to change the GPU from unity, something like init_parameters.sdk_gpu_id

Bvallon-sl, 2021-12-21:
Hi,

This parameter exists in Unity (https://github.com/stereolabs/zed-unity/blob/master/ZEDCamera/Assets/ZED/SDK/NativeInterface/ZEDCommon.cs#L1410) but not visible in the Inpector window.
You need to set it manually in the Awake() function of the ZEDManager.cs where the InitParameter struct is initialized.

I just realized this param was not used internally and I fixed it with the [last commit](https://github.com/stereolabs/zed-unity/commit/f2c4489c66ab538a5fe4d46e98d8e02a7b7bbc05) (on windows only, I'll fix it later today on linux).

Best regards,
Benjamin Vallon

AymanMukh, 2022-01-02:
Thank you @Bvallon-sl . After using the last commit file. I can now change the gpu in ZEDCommons.cs  
this.sdkGPUId = 2;

closed 2022-01-02 by AymanMukh