stereolabs/zed-sdk #454: Custom object detection and tracking with YOLOv5 s6 on custom image size

crdanfitz, 2022-01-04:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, feature request, or anything other than a bug report directly related to this project.

### Description

When following the [YOLOv5 with Zed custom box input example](https://github.com/stereolabs/zed-examples/tree/release_3.6/object%20detection/custom%20detector/cpp/tensorrt_yolov5), two problems are encountered when attempting to use a custom trained YOLOv5s6 model.

The first problem is that the generation of the .engine file fails:
`./yolov5_zed -s my_yolov5s6_model.wts my_yolov5s6_model.engine s6`
produces error:
> Loading weights: /home/dan/ChargeRobotics/custom_trained_yolo/yolov5/weights/crmodel.wts
> [01/03/2022-22:30:11] [E] [TRT] 3: [network.cpp::addScale::737] Error Code 3: API Usage Error (Parameter check failed at: optimizer/api/network.cpp::addScale::737, condition: shift.count > 0 ? (shift.values != nullptr) : (shift.values == nullptr))
> Segmentation fault (core dumped)

However, the same command succeeds on the same .wts on the latest version of [tensorrt](https://github.com/wang-xinyu/tensorrtx/tree/master/yolov5).

The second problem is that (using the .engine file produced by tensorrtx main), the _yolov5_zed_ detection mode initially crashes.
`./yolov5_zed -d my_yolov5s6_model.engine test_recording.svo`

> [ZED][Init] Depth mode: ULTRA

> CUDA error 700 at /usr/local/zed/zed-examples/object detection/custom detector/cpp/tensorrt_yolov5/src/yolov5.cpp:268CUDA error at /builds/sl/ZEDKit/lib/src/sl_zed/DetectionHandler.cpp:299 code=700(cudaErrorIllegalAddress) "cudaStreamWaitEvent(cu_strm, ev_image, 0)" 
> CUDA error at /builds/sl/ZEDKit/lib/src/sl_zed/DetectionHandler.cpp:311 code=700(cudaErrorIllegalAddress) "cudaEventRecord(ev_od, cu_strm)" 
> CUDA error at /builds/sl/ZEDKit/lib/src/sl_zed/DetectionHandler.cpp:67 code=700(cudaErrorIllegalAddress) "cudaMemcpy2DAsync(data.getPtr<T>(sl::MEM::CPU), data.getStepBytes(sl::MEM::CPU), data.getPtr<T>(sl::MEM::GPU), data.getStepBytes(sl::MEM::GPU), data.getWidthBytes(), data.getHeight(), cudaMemcpyDeviceToHost, strm)" 
> CUDA error at /builds/sl/ZEDKit/lib/src/sl_zed/DetectionHandler.cpp:67 code=700(cudaErrorIllegalAddress) "cudaMemcpy2DAsync(data.getPtr<T>(sl::MEM::CPU), data.getStepBytes(sl::MEM::CPU), data.getPtr<T>(sl::MEM::GPU), data.getStepBytes(sl::MEM::GPU), data.getWidthBytes(), data.getHeight(), cudaMemcpyDeviceToHost, strm)" 
> CUDA error at /builds/sl/ZEDKit/lib/src/sl_zed/DetectionHandler.cpp:344 code=700(cudaErrorIllegalAddress) "cudaStreamWaitEvent(cu_strm, ev_depth, 0)" 
> CUDA error at /builds/sl/ZEDKit/lib/src/sl_zed/DetectionHandler.cpp:67 code=700(cudaErrorIllegalAddress) "cudaMemcpy2DAsync(data.getPtr<T>(sl::MEM::CPU), data.getStepBytes(sl::MEM::CPU), data.getPtr<T>(sl::MEM::GPU), data.getStepBytes(sl::MEM::GPU), data.getWidthBytes(), data.getHeight(), cudaMemcpyDeviceToHost, strm)" 
> CUDA error at /builds/sl/ZEDKit/lib/src/sl_zed/DetectionHandler.cpp:387 code=700(cudaErrorIllegalAddress) "cudaStreamWaitEvent(cu_strm, ev_od, 0)" 
> in sl::ERROR_CODE sl::Mat::copyTo(sl::Mat&, sl::COPY_TYPE) const : cuda error [700]: an illegal memory access was encountered.
> in void sl::Mat::alloc(size_t, size_t, sl::MAT_TYPE, sl::MEM) : cuda error [700]: an illegal memory access was encountered.
> in sl::ERROR_CODE sl::Mat::setFrom(const sl::Mat&, sl::COPY_TYPE, cudaStream_t) : cuda error [12]: invalid pitch argument.
> [ZED][Grab] [WARNING] CUDA error returned:an illegal memory access was encountered ( 700 ) 
> in sl::ERROR_CODE sl::Mat::setFrom(const sl::Mat&, sl::COPY_TYPE, cudaStream_t) : cuda error [12]: invalid pitch argument.
> Error: (700): an illegal memory access was encountered

Although subsequent runs of the same command may succeed.

I suspect that these problems would be fixed by pulling from the [latest tensorrtx master](https://github.com/wang-xinyu/tensorrtx/tree/master/yolov5).

### Steps to Reproduce

1. Train a YOLOv5s6 model on a custom dataset according to the tutorial [here](https://docs.ultralytics.com/tutorials/train-custom-datasets/) with a command like `python train.py --img 1280 --data my_dataset.yaml --weights yolov5s6.pt`
2. Following Step 1 of the README.md: convert the resulting .pt pytorch model to a .wts weights file using gen_wts.py with a command like `python3 gen_wts.py --weights my_yolov5s6_model`
3. Set the INPUT_H and INPUT_W to both be `1280` in `yololayer.h`. 
4. Following the CONFIG step of the README.md: Set `CLASS_NUM` to be the nmber of classes in `my_dataset.yaml`.
4. Following step 2 of README.md: build the yolov5_zed executable: `make clean && make`
5. Following step 3 of README.md: (Attempt to) generate a TensorRT engine file from the resulting .wts file, with a command like: `./yolov5_zed -s my_yolov5s6_model.wts my_yolov5s6_model.engine s6`
6. Following step 4 of README.md: (Attempt to) run the sample on a .svo file or live camera with a command like: `./yolov5_zed -d my_yolov5s6_model.engine test_recording.svo`

### Expected Result

1. The pytorch .pt model can be successfully converted to a .wts model
2. The .wts file can be used fo generate a .engine file
3. The TensorRT engine can be used for custom model detection

### Actual Result

Generation of the TensorRT engine failes with error:
`./yolov5_zed -s my_yolov5s6_model.wts my_yolov5s6_model.engine s6`
produces error:
> Loading weights: /home/dan/ChargeRobotics/custom_trained_yolo/yolov5/weights/crmodel.wts
> [01/03/2022-22:30:11] [E] [TRT] 3: [network.cpp::addScale::737] Error Code 3: API Usage Error (Parameter check failed at: optimizer/api/network.cpp::addScale::737, condition: shift.count > 0 ? (shift.values != nullptr) : (shift.values == nullptr))
> Segmentation fault (core dumped)

As a workaround, the tensorrtx master branch can be used to generate this engine file, following the same instructions to build and use the yolov5 executable.

However, this engine file produces errors when used with the example yolov5_zed executable.
> [ZED][Init] Depth mode: ULTRA
[... 23 lines cut ...]
> OpenCV Error: Assertion failed (0 <= roi.x && 0 <= roi.width && roi.x + roi.width <= m.cols && 0 <= roi.y && 0 <= roi.height && roi.y + roi.height <= m.rows) in Mat, file /builds/sl/ZEDKit/lib/src/sl_core/opencv/core/matrix.cpp, line 507
> Error /builds/sl/ZEDKit/lib/src/sl_core/opencv/core/matrix.cpp:507: error: (-215) 0 <= roi.x && 0 <= roi.width && roi.x + roi.width <= m.cols && 0 <= roi.y && 0 <= roi.height && roi.y + roi.height <= m.rows in function Mat
> detect 

### ZED Camera model

ZED2i

### Environment

```shell
OS: Ubuntu 20.04
CPU: Xeon E3-150M
GPU: Quadro M3000M
Memory: 16GB
Zed SDK: 3.6.2
Cuda: 11.5
YOLOv5 model: s6, 10 classes, --img 1280
```

### Anything else?

The same results happen on the latest zed-exampled branch as on the latest zed sdk samples.

Suggest pulling from [tensorrtx master](https://github.com/wang-xinyu/tensorrtx/tree/master/yolov5).

adujardin, 2022-01-04:
Thanks for the report, it has been logged internally. Do you think you could share your model file (a dummy .pt model is possible) to gain time on our side to validate the resolution?

closed 2022-04-26 by github-actions[bot]

adujardin, 2022-05-11:
Just to let you know that we just pushed an update, [the sample is now compatible with yolov5 v6 models](https://github.com/stereolabs/zed-examples/tree/master/object%20detection/custom%20detector/cpp/tensorrt_yolov5_v6.0) and it should solve your issue