stereolabs/zed-sdk #458: Memory Leak in sdk sample of body tracking

juniniho, 2022-01-18:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, feature request, or anything other than a bug report directly related to this project.

### Description

There is a memory leak in sample of body tracking in C#.
The Visual Studio Diagnostic Tools show the memory leak.
Process Memory increase from 4GB to 4.7GB in 5 minutes. There were many times of GC performed, but the memory didn't decrease.
I build the sample in cpp also, it works well,with no memory leak.The bug only happens in csharp.

### Steps to Reproduce

1. Build sample under instruction in official site https://www.stereolabs.com/docs/app-development/csharp/use/#building-a-sample.
2. Open project in Visual Studio, run the project in release mode.
3. Observe the Process Memory In Diagnostic Tool or Windows Tasker Manager, You will find memory increases continuously.
...

### Expected Result

the process memory be stable in a small range.

### Actual Result

the process memory increases continuously.

### ZED Camera model

ZED2i

### Environment

```shell
OS:Windows 10 21H2
CPU:intel i5-9600KF
GPU:GTX 1060
ZED SDK VERSION:3.6.2 and 3.6.3
CUDA VERSION:11.5
CMAKE:3.21.1
Visual Studio:2019
Language: C#
```

Myzhar, 2022-01-18:
Thank you @juniniho 
we noted the issue and the team is going to work on a fix for it.
Stay tuned for the updates.

juniniho, 2022-01-18:
> Thank you @juniniho we noted the issue and the team is going to work on a fix for it. Stay tuned for the updates.

the memory leak also lead to heap corrupt which make my app crash.
After I downgrade ZED SDK to 3.5.2 & CUDA 11.1 ,the bug disappear. It works well in zed 3.5.2 with no memory leak.

Bvallon-sl, 2022-01-18:
Hi,

Thanks for the report.

The issue has been identified and will be fixed with the next 3.7 release.

Have a good day,

Best,
Benjamin V.

juniniho, 2022-03-30:
I upgrade zed sdk to 3.7.1, it runs perfect under gtx1660ti.
But the memory leak still exists under gtx1060, I think it maybe a bug of cuda driver for older video cards.

closed 2022-03-30 by juniniho