stereolabs/zed-sdk #461: Camera not detected

qhdwight, 2022-02-06:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, feature request, or anything other than a bug report directly related to this project.

### Description

ZED Diagnostic Camera Test - Camera not detected

```
{
    "Camera Test": {
        "ZEDCount": 0,
        "error": [
            "<b>Camera not detected</b> <br/> Make sure the camera is plugged in or try another USB 3.0 port."
        ]
    },
    "Graphics Card": {
        "deviceCount": 1,
        "deviceDriverVersion": 11060,
        "devices": [
            {
                "arch": "Pascal",
                "computeCapability": "6.1",
                "cores": 768,
                "name": "NVIDIA GeForce GTX 1050 Ti",
                "totalMemoryBytes": "4238802944",
                "totalMemoryMB": 4042.4375
            }
        ],
        "initResult": 0,
        "valid": [
            "<b>Graphics card: </b> NVIDIA GeForce GTX 1050 Ti"
        ]
    },
    "Processor": {
        "OS": "Debian GNU/Linux 11 ",
        "avxSupported": true,
        "coreCount": "8",
        "cpu": " Intel(R) Core(TM) i7-7700HQ CPU @ 2.80GHz",
        "datetime": "2022-02-05 06:56:18 PM",
        "hyperThreading": "1",
        "motherboard": "UX550VE, ASUSTeK COMPUTER INC.",
        "ramCapacity": "16.654041",
        "ramUsage": "",
        "ramUsed": "11.348529",
        "threadCount": "16",
        "valid": [
            "<b>Processor: </b>  Intel(R) Core(TM) i7-7700HQ CPU @ 2.80GHz",
            "<b>Motherboard: </b> UX550VE, ASUSTeK COMPUTER INC."
        ]
    },
    "USB": {
        "CorruptedFirmware": false,
        "USBList": [
            {
                "USB_path": "/2/2",
                "idProduct": "0x8153",
                "idVendor": "0x0bda"
            },
            {
                "USBMode": 3,
                "USB_path": "/2/1",
                "bDescriptorType": 1,
                "bDeviceProtocol": 1,
                "bLength": 18,
                "bMaxPacketSize0": 9,
                "bNumConfigurations": 1,
                "bcdDevice": "1.0",
                "bcdUSB": "3.0",
                "bcdUSBClass": 239,
                "bcdUSBSubClass": 2,
                "busNumber": 4,
                "device": "ZED",
                "iManufacturer": 1,
                "iProduct": 2,
                "iSerial": 0,
                "idProduct": "0xf582",
                "idVendor": "0x2b03"
            },
            {
                "USB_path": "/2",
                "idProduct": "0x0813",
                "idVendor": "0x2109"
            },
            {
                "USB_path": "/2",
                "idProduct": "0x2813",
                "idVendor": "0x2109"
            },
            {
                "USB_path": "/1",
                "idProduct": "0x1666",
                "idVendor": "0x0951"
            },
            {
                "USB_path": "/9",
                "idProduct": "0x0a2b",
                "idVendor": "0x8087"
            },
            {
                "USB_path": "/4",
                "idProduct": "0x5755",
                "idVendor": "0x13d3"
            },
            {
                "USB_path": "/11",
                "idProduct": "0x2544",
                "idVendor": "0x04f3"
            },
            {
                "USB_path": "/10",
                "idProduct": "0x0903",
                "idVendor": "0x04f3"
            }
        ],
        "USBMode": 3,
        "ZED Camera Module Detected": 62850,
        "ZEDDetected": true,
        "notTested": [
            "Please connect a camera to test the USB connection."
        ]
    },
    "ZED SDK": {
        "CUDA Toolkit version": "V11.5.119 -->located in /usr/local/cuda-11.5/bin/nvcc",
        "CUDA loading": false,
        "GPU Driver Version": "Driver Version                            : 510.47.03",
        "GPU Performance": "    Performance State                     : P8",
        "ZED SDK Version (Diag)": "3.6.4",
        "ZED SDK Version (RT)": "3.6.4",
        "ai": {
            "checkAI": true,
            "cudaVersion": 11050,
            "cudnnCudartVersion": 11050,
            "cudnnVersion": 8302,
            "cudnnVersionExpected": 8201,
            "tensorVersion": 8202
        },
        "binFiles": [
            "libsl_ai.so",
            "libsl_zed.so",
            "libsl_zed_static.a"
        ],
        "resourcesFiles": [
        ],
        "resourcesFilesExpected": [
            "objects_performance_2.1"
        ],
        "valid": [
            "<b>ZED SDK version:</b> 3.6.4",
            "<b>CUDA version:</b> V11.5.119"
        ]
    }
}
```

### Steps to Reproduce

I think this may be computer specific so I can't really say. Just install the SDK and plug in USB.

### Expected Result

Camera detected

### Actual Result

Camera not detected

### ZED Camera model

ZED

### Environment

```shell
Check .json text
```

### Anything else?

This also happens on Ubuntu 20.04

Myzhar, 2022-02-06:
The latest Ubuntu update upgraded the v4l2 driver and it changed the way it handles a parameter, hence breaking the connection with the ZED cameras.
The problem has been already fixed and we will release the patch v3.6.5 as soon as the QA tests are completed.
The fix will be available to download in the next few day.

qhdwight, 2022-02-06:
Awesome thank you!

obraun-sl, 2022-02-08:
ZED SDK v3.6.5 has been released and should fix this issue
--> https://www.stereolabs.com/developers/release/

KrasnovPavel, 2022-02-08:
Can confirm that issue fixed for me. Thanks!

closed 2022-02-08 by Myzhar

tkbala, 2023-06-20:
Hi,

I get the same issue with ZED cam 1 - CUDA 11.8, and RTX 4090. Any ideas how to fix it? I have pasted my diagnostic JSON below

```
{
    "AI Models": {
        "info": [
            "MULTI CLASS DETECTION is not optimized",
            "MULTI CLASS MEDIUM DETECTION is not optimized",
            "MULTI CLASS ACCURATE DETECTION is not optimized",
            "HUMAN BODY FAST DETECTION is not optimized",
            "HUMAN BODY MEDIUM DETECTION is not optimized",
            "HUMAN BODY ACCURATE DETECTION is not optimized",
            "HUMAN BODY 38 FAST DETECTION is not optimized",
            "HUMAN BODY 38 MEDIUM DETECTION is not optimized",
            "HUMAN BODY 38 ACCURATE DETECTION is not optimized",
            "HUMAN BODY 70 FAST DETECTION is not optimized",
            "HUMAN BODY 70 MEDIUM DETECTION is not optimized",
            "HUMAN BODY 70 ACCURATE DETECTION is not optimized",
            "PERSON HEAD DETECTION is not optimized",
            "PERSON HEAD ACCURATE DETECTION is not optimized",
            "REID ASSOCIATION is not optimized",
            "NEURAL DEPTH is not optimized"
        ]
    },
    "Camera Test": {
        "InternalDevicesCount": 1,
        "ZEDCount": 0,
        "error": [
            "<b>Camera not detected</b> <br/> Make sure the camera is plugged in or try another USB 3.0 port."
        ]
    },
    "Devices": {
        "ASMediaFound": false,
        "CorruptedFirmware": false,
        "OC_Cam_Detected": false,
        "OtherUSBDevices": [
        ],
        "USBList": [
            {
                "USB_path": "/11",
                "idProduct": "0x0608",
                "idVendor": "0x05e3"
            },
            {
                "USB_path": "/13",
                "idProduct": "0x0608",
                "idVendor": "0x05e3"
            },
            {
                "USB_path": "/14",
                "idProduct": "0x0aaa",
                "idVendor": "0x8087"
            },
            {
                "USBMode": 3,
                "USB_path": "/23",
                "bDescriptorType": 1,
                "bDeviceProtocol": 1,
                "bLength": 18,
                "bMaxPacketSize0": 9,
                "bNumConfigurations": 1,
                "bcdDevice": "1.0",
                "bcdUSB": "3.0",
                "bcdUSBClass": 239,
                "bcdUSBSubClass": 2,
                "busNumber": 1,
                "device": "ZED",
                "iManufacturer": 1,
[... 218 lines cut ...]
            "nvonnxparser.dll",
            "nvparsers.dll",
            "sl_ai64.dll",
            "sl_zed64.dll",
            "zlibwapi.dll"
        ],
        "driverVersion": "545.29",
        "drivervDllVersion": "31.0.15.4529",
        "resourcesFiles": [
        ],
        "resourcesFilesExpected": [
            "objects_performance_3.0"
        ],
        "runtimeDllVersion": 11080,
        "valid": [
            "<b>ZED SDK version:</b> 4.0.3",
            "<b>CUDA version:</b> V11.8.89"
        ],
        "warning": [
            "Multiple version of CUDA has been detected. It is recommended to uninstall other versions than 11.8."
        ]
    }
}

```

lixuexian0724, 2023-12-26:
[ZED_Diagnostic_Results.json](https://github.com/stereolabs/zed-sdk/files/13769703/ZED_Diagnostic_Results.json)
I also encountered a similar problem. The diagnosis results have been uploaded. Can anyone help me answer it?
[image]

ssht428, 2024-07-21:
Hi,

 I buy an used zed mini from other guy, but he don't have the usb-cable, so now i tried several usb cables(type-c to c, c to a...), the camera can't work on mine computer,
I already installed ZED_SDK_Windows_cuda11.8_v4.1.3.exe  and cuda 11.8 drvier on windows 10, the camera is recognized as HID input device, so I want to ask:
1. do i have to buy the original usb-cable from the vendor? does this zed mini camera support normal usb-cable?
2. how do i changed the driver from HID to zed camera?
3. what should i do to get this camera work?
[image]
[image]
[image]
[image]
[ZED_Diagnostic_Results.json](https://github.com/user-attachments/files/16323841/ZED_Diagnostic_Results.json)

Looking for your reply, thanks.

tejasmk-tkp, 2025-10-06:
> Hi,
>
> I buy an used zed mini from other guy, but he don't have the usb-cable, so now i tried several usb cables(type-c to c, c to a...), the camera can't work on mine computer, I already installed [...]

Facing the same issue even with cable given by ZED, it's being detected as an HID port

ssht428, 2025-10-07:
> Hi,
>
> I buy an used zed mini from other guy, but he don't have the usb-cable, so now i tried several usb cables(type-c to c, c to a...), the camera can't work on mine computer, I already installed [...]

> 
> Facing the same issue even with cable given by ZED, it's being detected as an HID port 

 _Originally posted by @tejasmk-tkp in [#461](https://github.com/stereolabs/zed-sdk/issues/461#issuecomment-3372329188)_

-----------------------------------------
I buy a compatible usb-c cable and try both two side of the type-c interface, only one side work,the other side will only be reconized as HID.
I can't remember the address of which blog said that zed use typec pins on bothside, not as the standard way.

[image]