stereolabs/zed-sdk #462: Mac OS

shlomi-amitai, 2022-02-08:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, feature request, or anything other than a bug report directly related to this project.

### Description

Do you have a MAC OS supported version?

### Steps to Reproduce

1.
2.
3.
...

running on Mac

### Expected Result

working properly

### Actual Result

Not working 

### ZED Camera model

ZED2i

### Environment

```shell
Mac OS
```

obraun-sl, 2022-02-10:
Hi,
Sorry but it is not planned to support Mac OS X at the moment.

shlomi-amitai, 2022-02-10:
What about recording only?

closed 2022-04-26 by github-actions[bot]

rajkundu, 2025-09-10:
In case it is useful to anyone else:

I wanted to use ZED 2i as a simple stereo camera on macOS via UVC. I found Stereolabs' [zed-opencv-native](https://github.com/stereolabs/zed-opencv-native) repo which does just that, but it requires the device serial number to download the calibration files.
I didn't have access to my device's box, and the SDK/Explorer app doesn't exist for macOS, but I was able to get my device's serial number using the command `ioreg -p IOUSB -l -w 0 | grep -E "ZED|Serial"` (courtesy of ChatGPT), under `"kUSBSerialNumberString"` in the output.