stereolabs/zed-sdk #463: OpenCV Error: Assertion failed (0 <= roi.x && 0 <= roi.width && roi.x + roi.width <= m.cols && 0 <= roi.y && 0 <= roi.height && roi.y + roi.height <= m.rows) in Mat, file /builds/sl/ZEDKit/lib/src/sl_core/opencv/core/matrix.cpp, line 508 Segmentation fault (core dumped)

ASingh245, 2022-02-12:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, feature request, or anything other than a bug report directly related to this project.

### Description

I am using the ZED2 Camera on Jetson Xavier AGX. The ZED SDK is installed for Jetson Jetpack 4.6. When I try to execute the opencv_dnn_yolov4 from https://github.com/stereolabs/zed-examples/tree/master/object%20detection/custom%20detector/cpp/opencv_dnn_yolov4. The camera opens for 1 second and closes after that with the error mentioned as:  
OpenCV Error: Assertion failed (0 <= roi.x && 0 <= roi.width && roi.x + roi.width <= m.cols && 0 <= roi.y && 0 <= roi.height && roi.y + roi.height <= m.rows) in Mat, file /builds/sl/ZEDKit/lib/src/sl_core/opencv/core/matrix.cpp, line 508
Segmentation fault (core dumped).

Please let me know how can I fix this error.

### Steps to Reproduce

1. Clone the repo https://github.com/stereolabs/zed-examples
2. Go to folder zed-examples/object detection/custom detector/cpp/opencv_dnn_yolov4
3. mkdir build
cd build
cmake ..
make
cd ../
4. ./build/opencv_dnn_zed

### Expected Result

model should run unintereptedly.

### Actual Result

OpenCV Error: Assertion failed (0 <= roi.x && 0 <= roi.width && roi.x + roi.width <= m.cols && 0 <= roi.y && 0 <= roi.height && roi.y + roi.height <= m.rows) in Mat, file /builds/sl/ZEDKit/lib/src/sl_core/opencv/core/matrix.cpp, line 508
Segmentation fault (core dumped).

### ZED Camera model

ZED2

### Environment

```shell
OS: Ubuntu LTS 18.04, bionic
ZED SDK Version: ZED_SDK_Tegra_JP46_v3.6.5
CUDA Version: 10.2
```

br0tda, 2022-02-28:
Same problem with custom detector in SDK 3.7

br0tda, 2022-02-28:
> Same problem with custom detector in SDK 3.7

Fixed adding in cv::Rect **get_rect**(cv::Mat& img, float bbox[4]) in **common.hpp** these lines

```
int r_minus_l = r - l;
int b_minus_t = b - t;

    if(l <0 )
        l = 0;
    if(t < 0 )
        t = 0; 
    if(r_minus_l < 0 )
        r_minus_l = 0;
    if(b_minus_t < 0 )
        b_minus_t = 0;

    return cv::Rect(l, t, r_minus_l, b_minus_t); 
```

ASingh245, 2022-03-01:
Ok, I will try thanks

closed 2022-04-26 by github-actions[bot]