stereolabs/zed-sdk #466: C# sample doesn't work.

hagsac, 2022-02-25:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, feature request, or anything other than a bug report directly related to this project.

### Description

"nuget 3.7 is not installed" is displayed.
I changed it to 3.6 but the execution failed.

[image]

### Steps to Reproduce

1. vs2022
2. sdj 3.7
3. cmake install
4. nuget error
...

### Expected Result

normal behavior

### Actual Result

[image]

### ZED Camera model

ZED2i

### Environment

```shell
win 10 pro
1660 ti
16gb ram
sdj 3.7
example c#
```

Bvallon-sl, 2022-02-25:
Hi, 
the nuget package for 3.7 is not available yet.

It will be published today.

Best.
Benjamin V.

Bvallon-sl, 2022-02-25:
It has now been published and will be available soon after validation.
Do not hesitate to contact us if you still have issues using the nuget package.

Best regards,
Benjamin Vallon

closed 2022-04-26 by github-actions[bot]