stereolabs/zed-sdk #468: sl::Camera::getDeviceList() returns 0, and no camera present with ZED_Explorer

Rajesh-VR, 2022-03-03:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, feature request, or anything other than a bug report directly related to this project.

### Description

I am connecting my zed2i camera into my pc (Ubuntu20.04), and try to call "sl::Camera::getDeviceList()" (it is with my c++ project). It always returns zero, and then I tried to run "ZED_Explorer" also, but there is nothing showing with "ZED_Explorer".

sample code is like:
    const auto devList = sl::Camera::getDeviceList();

    if (devList.empty()) {
        throw Exception("No camera present");  // always hitting here
    }

### Steps to Reproduce

1. Write simple program to get the sl::camera list
2. call like "const auto devList = sl::Camera::getDeviceList();"
3. check the device count

### Expected Result

sl::Camera::getDeviceList() call should return the proper number of zed2i camera count, and ZED_Explorer should work with the connected zed2i cameras.
[image]

### Actual Result

1. "sl::Camera::getDeviceList()" call return 0, even if there is one zed2i camera already connected.
2. ZED_Explorer is detecting the connected zed2i camera

### ZED Camera model

ZED2i

### Environment

```shell
OS: Ubuntu20.04
ZED SDK version: 3.5.1
```

Myzhar, 2022-03-03:
Hi @Rajesh-VR 
please install the latest ZED SDK:
https://www.stereolabs.com/developers/release/

A recent Ubuntu upgrade introduced a modification in the v4l2 kernel driver that broke the connection between UVC and the ZED SDK. We fixed the problem starting from ZED SDK v3.6.5.

I suggest you get the latest ZED SDK v3.7 to exploit the new NEURAL depth mode, something that has never been seen on other RGBD devices: https://www.stereolabs.com/developers/release/

Rajesh-VR, 2022-03-10:
Thanks @Myzhar 
It worked !!

closed 2022-03-10 by Rajesh-VR