stereolabs/zed-sdk #469: NEURAL depth mode not working

pezosanta, 2022-03-08:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, feature request, or anything other than a bug report directly related to this project.

### Description

To record data for an ML project, we (need to) use nvidia accelerated laptops (Windows 10). As your SDK 3.7 came out we decided to test the brand new NEURAL depth mode, however we can not make it to work. 

According to the following SDK 3.7 release notes statement:
_Use `ZED_Diagnostic -nrlo` after installing the ZED SDK or launch your application with the new depth mode and wait for the optimization to finish._
we need to optimize the NEURAL depth DL model by either running the ZED Diagnostic tool or running our application with using NEURAL depth mode.

So first, we ran the ZED Diagnostic app which said that _NEURAL DEPTH is not optimized_. After that we ran our application with NEURAL depth mode, the app started immediately and got fully black depth maps (with retreive_image). Therefore, for some reason the DL model optimiziation is not executed at all.

### Steps to Reproduce

Run the following code:

```
import sys
import cv2
import pyzed.sl as sl

camera = sl.Camera()

init_params = sl.InitParameters()
init_params.camera_resolution = sl.RESOLUTION.HD1080
init_params.camera_fps = 15
init_params.depth_mode = sl.DEPTH_MODE.NEURAL
init_params.coordinate_units = sl.UNIT.MILLIMETER
init_params.depth_minimum_distance = 1000
init_params.depth_maximum_distance = 40000
init_params.camera_image_flip = sl.FLIP_MODE.AUTO
init_params.depth_stabilization = True
runtime_params = sl.RuntimeParameters(confidence_threshold=50, texture_confidence_threshold=100)

err = camera.open(init_params)
if err != sl.ERROR_CODE.SUCCESS:
    print(err)
    sys.exit()

image = sl.Mat()
depth_map = sl.Mat()

while True:
    if camera.grab(runtime_params) == sl.ERROR_CODE.SUCCESS:
        camera.retrieve_image(image, sl.VIEW.LEFT) # Retrieve left image
        camera.retrieve_image(depth_map, sl.VIEW.DEPTH) # Retrieve depth

        numpy_image = image.get_data()
        numpy_depth_map = depth_map.get_data()

        cv2.imshow('DEPTH', numpy_depth_map)
        if cv2.waitKey(1) & 0xFF == ord('q'):
            break
```

### Expected Result

The room looks like this:
[image]

The following depth map was taken using ULTRA depth mode:
[image]

### Actual Result

With NEURAL depth mode the depth mode looks like this:
[image]

Furthermore, with the following code, we counted the occurance of unique values in the retreived numpy depth maps:
`unique, counts = np.unique(numpy_depth_map, return_counts=True)`
The result was: [ 0 255] [6220800 2073600], which means that only 0 and 255 values are in the depth map.

However, during the application run, the python process (that runs the ZED SDK with NEURAL depth mode) uses 377MiB of GPU ram with a 20-55% GPU utilization (based on nvidia-smi) and around 30% CPU utilization (based on the Task Manager).

### ZED Camera model

ZED2i

### Environment

```shell
OS: Windows 10
CPU: Intel Core i7-8565U @ 1.80 GHz
GPU: Nvidia GeForce MX250 @ 2GB
Nvidia driver version: 497.29
CUDA version: 11.5
Camera firmware: 1523
SDK version: 3.7
Python verson: 3.8.10
```

Myzhar, 2022-03-08:
There is a bug in `ZED_Diagnostic ` so the flag `-nrlo` is not working as expected. 
Try to optimize the Neural Depth model by using the workaround command
`ZED_Diagnostic -ais 8`

pezosanta, 2022-03-08:
Maybe it is a dumb question, but how (where) should I enter this command in Windows 10? (The terminal does not recognize it as an internal or  external command.)

pezosanta, 2022-03-08:
Hi @Myzhar, thanks for your quick reply!

I used the workaround command you have just suggested and it seems like the NEURAL DEPTH model is optimized now (according to the ZED Diagnostic).
`Optimizing: NEURAL DEPTH...                                                                                              |[ZED][INFO] AI model not found, downloading                                                                           \Done                                                                                                                   [ZED][INFO] Done                                                                                                         |[ZED][INFO] Please wait while the AI model is being optimized for your graphics card                                   This operation will be run only once and may take a few minutes                                                        [ZED][INFO] Done `

[image]

However, running my application with NEURAL DEPTH still retreives "black" depth maps with only 0 and 255 values in it.

Myzhar, 2022-03-09:
Does ZED Depth Viewer work correctly?

pezosanta, 2022-03-09:
@Myzhar yes, ZED Depth Viewer works correctly with NEURAL depth mode. But still, the python script (detailed above) retrieves black depth map images with only 0 and 255 values in it.

[image]

Myzhar, 2022-03-09:
Hi @pezosanta 
I tested your Python script under Ubuntu and I'm facing the same issue.
We are debugging the Python wrapper searching for a solution.
Stay tuned.

pezosanta, 2022-03-18:
Hi @Myzhar, any updates so far?

Myzhar, 2022-03-18:
@pezosanta the bug will be fixed in the oncoming ZED SDK v3.7.1 that brings other fixes to the NEURAL depth mode.

pezosanta, 2022-03-18:
@Myzhar thanks for the update. When will be the new SDK version released? (We start the training data recording in the upcoming days and I need to know whether the NEURAL depth mode will be ready or not.)

pezosanta, 2022-03-28:
Hi @Myzhar, I have just tested the new ZED SDK 3.7.1 on Windows 10 and on Jetson Xavier NX. On the Jetson Xavier NX, the python wrapper seems to work correctly with the NEURAL depth mode, however on Windows 10 it still retrieves black depth map images with only 0 and 255 values in the numpy array. Could you confirm this behaviour on windows 10?

Myzhar, 2022-03-29:
Hi @pezosanta 
we just tested on Windows and everything seems to work correctly. Have you verified to have correctly updated the Python API?

alassagne-sl, 2022-03-29:
Hello, there might be an issue with the optimization of the model. Can you try to restore it from scratch ?
To do that,
- Go to C:\ProgramData\Stereolabs\resources
- Remove the file neural_depth_*.model and  neural_depth_*.model_optimized_*
- Launch your script again
On windows, the model optimization will probably take half an hour. In the meantime, please try not to use your GPU, and don't launch any Zed program (or anything else that might use TensorRT).
I hope that will work :)

Myzhar, 2022-03-29:
What @alassagne-sl says is correct.
Please delete all the AI models in the ZED SDK installation folder, then run the command `ZED Diagnostic.exe --aio` from command line.
It will download and optimize all the AI models.

alassagne-sl, 2022-03-30:
Beware that, on windows, optimizing all the models will take hours. It's something we're working on since the time is 10 times shorter on linux. You can also use ZED_Diagnostics -ais X to optimize only one model.
Optimizations can be done through ZED Diagnostics, ZED Depth Viewer, or directly on the fly if the SDK detects that it needs an optimization. Optimization must be updated for new versions of the SDK, of CUDA, changes of hardware.

Still, it's very weird that the optimization fails. What's your GPU ? CUDA version ? Can you send us the output of ZED Diagnostics ?

alassagne-sl, 2022-03-30:
Thanks for this output. Also, sorry that I forgot that you gave us your GPU and cuda version earlier. We'll investigate and keep in touch.

alassagne-sl, 2022-03-31:
Hello, that seems more like it.
Let's start over a little. Since you are using a virtualenv. I tried the same setup on my machine.
I encountered the same dll issues as you and, when I copied the 2 dlls from C:\Program Files (x86)\ZED SDK\bin it worked.
We fixed a bug in 3.7.1 which looks so much close to yours that it'd be strange if it was another.

Here are a few things you can try to help us find the issue, or even to solve it :

- What happens without the python virtual environment ?
- Try to place the zed in a very different environment (or with an SVO saved from somewhere else). The bug I mention was affected by the environment. That
- Try to launch not to retrieve  sl.VIEW.DEPTH but sl.MEASURE.XYZRGBA. What happens ? You can take inspiration in samples if you want to display that with opengl
- When you cleaned the whole resources folders that contains the optimization files and you launched your script (not depth viewer) what happened ? This is very important. If your script does not run any model optim, it means that it finds an optimized model somewhere. That means that your cudnn/tensorrt installation may be broken. If you're brave enough to reinstall CUDA completely, that would solve that kind of issue.

closed 2022-05-07 by github-actions[bot]

GoncaloGarganta, 2024-02-01:
Hi, I was wondering if this problem was resolved. I seem to have encountered a similar problem if not the same one.

When I open ZED Depth Viewer, everything works fine, as can be seen in the image below. I am using the neural depth mode and the image appears with no problem. However, when I go to the settings and try to save the depth image as a PNG it saves a completely black image. This does not happen in Ultra depth mode ( the image is saved correctly).

[image]

Moreover, I am using the zed-opencv script to capture the image from the camera, and by using "init.depth_mode = sl.DEPTH_MODE.ULTRA" everything works fine, I can visualize the depth image and save it. 
When I change it to "init.depth_mode = sl.DEPTH_MODE.NEURAL" then the depth image becomes all black again, despite the compiler not giving any errors. 

I understand there is a separate issue thread for the zed-opencv script, but since my problem is happening in the depth viewer app as well I believe this is the correct place to ask.

I have uninstalled the SDK and reinstalled again, checked ZED diagnostics and the NEURAL depth mode is optimized, I apologize if I missed something, but any help would be appreciated!

This is my environment: 

OS: Windows 11
CPU: 13th Gen Intel(R) Core(TM) i7-13700HX   2.10 GHz
GPU: Nvidia GeForce RTX 4060 Laptop GPU
Nvidia driver version: 546.26
CUDA version: 12.1
Camera firmware: 1523
SDK version: 4.0.8
Python verson: 3.11.7

Myzhar, 2024-02-01:
@GoncaloGarganta the issue has been noted and the SDK Team will try to reproduce and eventually fix it.

braoutch, 2024-02-01:
Hi, are you in a virtual environment?

alassagne-sl, 2024-02-01:
Hi, are you using a virtualenv?

GoncaloGarganta, 2024-02-01:
Hello, no I am not using a virtualenv.