stereolabs/zed-sdk #470: object detection segmentation fault ZED sdk 3.7 cuda 10.2

Murdism, 2022-03-11:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, feature request, or anything other than a bug report directly related to this project.

### Description

Hello i have the similar issue discussed [here](https://github.com/stereolabs/zed-examples/issues/436#issuecomment-953170936) but with sdk 3.7.0, 

i am using
# ubuntu 18.04 with cuda 10.2 and ZED sdk 3.7.0 GPU RTX 2080 with 12GB
# object detection example causes segmentation fault. ->  Ai optimization

 

### Steps to Reproduce

1. install sdk 3.70 , cuda 10.2 for linux
2. run a simple example with camera -> works fine
3. object detection causes segmentation error
...

### Expected Result

Normal operation.

### Actual Result

-> here is a result of ->  $ /usr/local/zed/tools/ZED_Diagnostic -aio

Optimizing all AI models
Optimizing: MULTI CLASS DETECTION...
  [ZED][INFO] AI model not found, downloading
\[ZED][INFO] Done
[ZED][INFO] Done
 |[ZED][INFO] Please wait while the AI model is being optimized for your graphics card
 This operation will be run only once and may take a few minutes 
|Stack trace (most recent call last):
#11   Object "[0xffffffffffffffff]", at 0xffffffffffffffff, in 
#10   Object "/usr/local/zed/tools/ZED_Diagnostic", at 0x432ef9, in 
#9    Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7f2805237c86, in __libc_start_main
#8    Object "/usr/local/zed/tools/ZED_Diagnostic", at 0x41bc8d, in 
#7    Object "/usr/local/zed/tools/ZED_Diagnostic", at 0x434b35, in 
#6    Object "/usr/local/zed/tools/ZED_Diagnostic", at 0x46d3f2, in 
#5    Object "/usr/local/zed/tools/ZED_Diagnostic", at 0x46c9ca, in 
#4    Object "/usr/local/zed/lib/libsl_ai.so", at 0x7f27febfa827, in 
#3    Object "/usr/local/zed/lib/libsl_ai.so", at 0x7f27febfdf30, in 
#2    Object "/usr/local/zed/lib/libsl_ai.so", at 0x7f27fec3166e, in 
#1    Object "/usr/local/zed/lib/libsl_ai.so", at 0x7f27fec2dc2a, in 
#0    Object "/usr/local/zed/lib/libsl_ai.so", at 0x7f27fec29a34, in 
Segmentation fault (Address not mapped to object [0x8])
Segmentation fault (core dumped)

Running $ ldd /usr/local/zed/sl_ai.so shows:
ldd: /usr/local/zed/sl_ai.so: No such file or directory

### ZED Camera model

ZED2

### Environment

```shell
OS: ubuntu 18.04
CPU: Intel x86 64bit
GPU : RTX 2080 Ti 12GB
ZED SDK: 3.7.0 cuda 10.2
```

obraun-sl, 2022-03-14:
Hi,

Can you run : 
$ ldd /usr/local/zed/lib/libsl_ai.so 

(This is the correct path)

Murdism, 2022-03-14:
> Hi,
> 
> Can you run : $ ldd /usr/local/zed/lib/libsl_ai.so
> 
> (This is the correct path)

Can you run :
$ ldd /usr/local/zed/lib/libsl_ai.so

Murdism, 2022-03-14:
> Hi,
> 
> Can you run : $ ldd /usr/local/zed/lib/libsl_ai.so
> 
> (This is the correct path)

	linux-vdso.so.1 (0x00007ffde1ab2000)
	libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007fc6cbf43000)
	libdl.so.2 => /lib/x86_64-linux-gnu/libdl.so.2 (0x00007fc6cbd3f000)
	librt.so.1 => /lib/x86_64-linux-gnu/librt.so.1 (0x00007fc6cbb37000)
	libpng16.so.16 => /usr/lib/x86_64-linux-gnu/libpng16.so.16 (0x00007fc6cb905000)
	libz.so.1 => /lib/x86_64-linux-gnu/libz.so.1 (0x00007fc6cb6e8000)
	libjpeg.so.8 => /usr/lib/x86_64-linux-gnu/libjpeg.so.8 (0x00007fc6cb480000)
	libcuda.so.1 => /usr/lib/x86_64-linux-gnu/libcuda.so.1 (0x00007fc6ca25f000)
	libnvinfer.so.8 => /usr/local/cuda-10.2/lib64/libnvinfer.so.8 (0x00007fc6bc938000)
	libnvinfer_plugin.so.8 => /usr/local/cuda-10.2/lib64/libnvinfer_plugin.so.8 (0x00007fc6bb300000)
	libnvonnxparser.so.8 => /usr/local/cuda-10.2/lib64/libnvonnxparser.so.8 (0x00007fc6bae57000)
	libarchive.so.13 => /usr/lib/x86_64-linux-gnu/libarchive.so.13 (0x00007fc6baba7000)
	libsl_zed.so => /usr/local/zed/lib/libsl_zed.so (0x00007fc6b19de000)
	libstdc++.so.6 => /usr/lib/x86_64-linux-gnu/libstdc++.so.6 (0x00007fc6b1655000)
	libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007fc6b12b7000)
	libgomp.so.1 => /usr/lib/x86_64-linux-gnu/libgomp.so.1 (0x00007fc6b1088000)
	libgcc_s.so.1 => /lib/x86_64-linux-gnu/libgcc_s.so.1 (0x00007fc6b0e70000)
	libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007fc6b0a7f000)
	/lib64/ld-linux-x86-64.so.2 (0x00007fc6cce65000)
	libcudnn.so.8 => /usr/local/cuda-10.2/lib64/libcudnn.so.8 (0x00007fc6b0855000)
	libcublas.so.10 => /usr/lib/x86_64-linux-gnu/libcublas.so.10 (0x00007fc6ac59f000)
	libcublasLt.so.10 => /usr/lib/x86_64-linux-gnu/libcublasLt.so.10 (0x00007fc6aa70c000)
	libnettle.so.6 => /usr/lib/x86_64-linux-gnu/libnettle.so.6 (0x00007fc6aa4d6000)
	libacl.so.1 => /lib/x86_64-linux-gnu/libacl.so.1 (0x00007fc6aa2ce000)
	liblzo2.so.2 => /lib/x86_64-linux-gnu/liblzo2.so.2 (0x00007fc6aa0ac000)
	liblzma.so.5 => /lib/x86_64-linux-gnu/liblzma.so.5 (0x00007fc6a9e86000)
	liblz4.so.1 => /usr/lib/x86_64-linux-gnu/liblz4.so.1 (0x00007fc6a9c6a000)
	libbz2.so.1.0 => /lib/x86_64-linux-gnu/libbz2.so.1.0 (0x00007fc6a9a5a000)
	libxml2.so.2 => /usr/lib/x86_64-linux-gnu/libxml2.so.2 (0x00007fc6a9699000)
	libturbojpeg.so.0 => /usr/lib/x86_64-linux-gnu/libturbojpeg.so.0 (0x00007fc6a9429000)
	libusb-1.0.so.0 => /lib/x86_64-linux-gnu/libusb-1.0.so.0 (0x00007fc6a9211000)
	libnvcuvid.so.1 => /usr/lib/x86_64-linux-gnu/libnvcuvid.so.1 (0x00007fc6a8cfd000)
	libnvidia-encode.so.1 => /usr/lib/x86_64-linux-gnu/libnvidia-encode.so.1 (0x00007fc6a8ae4000)
	libattr.so.1 => /lib/x86_64-linux-gnu/libattr.so.1 (0x00007fc6a88df000)
	libicuuc.so.60 => /usr/lib/x86_64-linux-gnu/libicuuc.so.60 (0x00007fc6a8527000)
	libudev.so.1 => /lib/x86_64-linux-gnu/libudev.so.1 (0x00007fc6a8309000)
	libicudata.so.60 => /usr/lib/x86_64-linux-gnu/libicudata.so.60 (0x00007fc6a6760000)

closed 2022-04-26 by github-actions[bot]