stereolabs/zed-sdk #471: NEURAL Depth mode supported?

AndreV84, 2022-03-13:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, bug report, or anything other than a feature request directly related to this project.

### Proposal

add NEURAL Depth mode

### Use-Case

save more detailed 3d map processed by NEURAL mode

Myzhar, 2022-03-13:
@AndreV84 can you please add more details?

Myzhar, 2022-03-13:
You can set depth mode to NEURAL by changing the relative depth mode parameter:
https://github.com/stereolabs/zed-examples/blob/master/spatial%20mapping/advanced%20point%20cloud%20mapping/cpp/src/main.cpp#L46

AndreV84, 2022-03-13:
@Myzhar 
Thank you for your reply
Is there a chance to just add one line to the basic example code?
after the line https://github.com/stereolabs/zed-examples/blob/ed3f068301fbdf3898f7c42de864dc578467e061/spatial%20mapping/basic/cpp/src/main.cpp#L46 
?
``` 
init_parameters.depth_mode = DEPTH_MODE::NEURAL;

Myzhar, 2022-03-13:
It's an example code, you can adapt it to your requirements without problems.

AndreV84, 2022-03-13:
so lets close the issue

closed 2022-03-13 by AndreV84