stereolabs/zed-sdk #472: Having an instance of sl::Camera open messes with CUDA Memory

PNeigel, 2022-03-15:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, feature request, or anything other than a bug report directly related to this project.

### Description

I created a neural network model using LibTorch (c++ library of pytorch) that  does some computations on the GPU. That model works perfectly fine if I load images in my main loop (in my case with opencv) and feed them to the network.

But if I instead use an instance of sl::Camera to load images from an .svo file, my program regularly crashes with weird CUDA memory exceptions. After debugging for hours and hours I realized that in many Torch arrays the values are almost randomly flipping. It seems like "another process" is writing into the addresses used up by these tensors.
Now I realized that if I preload all the rgb images and depth measures in the svo, close the cl::Camera instance with .close() and loop over the cached images, everything works fine again. So my conclusion is that having an sl::Camera instance open somehow messes with the CUDA memory allocated by the torch::Tensors.

Now of course caching all images takes up a lot of memory. Is there any way to not do that and simply grab all images in my loop "live" from the sl::Camera instance?

Am I right in my assumption? Am I doing something wrong? Am I not following some best practice guidelines?

### Steps to Reproduce

1. Create a LibTorch model on GPU
2. Grab an image from sl::Camera
3. Feed that image to the LibTorch model while the sl::Camera instance is open

### Expected Result

CUDA memory allocated for torch::Tensors is not touched by the ZED SDK at all.

### Actual Result

CUDA memory allocated for torch::Tensors is "externally" manipulated while sl::Camera is open.

### ZED Camera model

ZED2i

### Environment

```shell
System: Windows 10
CPU: AMD Ryzen 9 5900x
GPU: Nvidia RTX 3090
ZED SDK: 3.6.5
LibTorch: 1.10.2+cu113
CUDA: 11.3
IDE: Visual Studo 17
```

closed 2022-04-26 by github-actions[bot]

Saligz, 2024-11-29:
I have the same issue as you are describing, did you found a solution ?

iskanderuse, 2025-04-22:
When we have in one application zed camera zed and pytorch at the same time, then we have error: "No NVIDIA graphics card detected. Install an NVIDIA GPU, CUDA and restart your computer after completing installation". 
Although it is not so, strange behavior for the camera...